liblogicalaccess/liblogicalaccess #73: Initialize a reader connection failed

nimch, 2015-01-06:
Hello.
I'm a developer and I'm trying to use liblogicalaccess. I work in a virtualized debian 8 (vmware), I installed dependencies.

I've started with the examples on the wiki page and I'm stuck on the first : "Initialize a reader connection". Compilation is going well but the program crashes at the 3rd line because readerConfig-> getReaderProvider () is NULL. I read in the FAQ to verify the installation of pcsc-lite, it's done...

I have two readers: Gemalto Prox-DU and Omnikey 5321. Same behavior with each device.

Additional info, I've tested a python program that uses pyscard which also uses pcsc-lite and it worked. Python programme also worked on my host (windows seven).
Do you have an idea for me to go further ?

Thank you for your attention
Best Regards

PS : First I tried on windows but I could not solve link error issues with the lib (__declspec(dllimport) public: static class logicalaccess::LibraryManager * __cdecl logicalaccess::LibraryManager::getInstance(void)" bla bla bla).

Liryna, 2015-01-06:
Hi,
it happened that readerConfig->getReaderProvider () is NULL when the project is not linking with pcsc-lite library.
Can you try to add pcsc-lite as dependancy to your project ?

For Windows, I have never seen such error :'( logicalaccess::LibraryManager::getInstance is part of the Core so there should be no problem
could you give more information ?

nimch, 2015-01-06:
Hi
I've explicitly linked the lib with : g++ -g main.cpp -llogicalaccess -lpcsclite -std=c++11 -DUNIX -o prog and I had the same behaviour.

For windows, I compiled with Visual Studio Express 2013 and added liblogicalaccess path to the linker.

Thanks !

Liryna, 2015-01-06:
Quick test: Can you try to do g++ -g main.cpp -lpcsclite -std=c++11 -llogicalaccess -DUNIX -o prog

nimch, 2015-01-06:
same behaviour ! segmentation fault !

Liryna, 2015-01-06:
Ok I probably understood your probleme.
Liblogicalaccess is using the config file (https://github.com/islog/liblogicalaccess/blob/master/liblogicalaccess.config) for getting the PluginFolders.
This config file have to be in the same folder as the software for linux and in the same folder as the library for windows.

PluginFolders is where you have libpcscreaders.so

nimch, 2015-01-06:
Yes you do !

After modifying PluginFolders' path (was /usr/lib/logicalaccess, set to /usr/lib) my reader seems to be initialized correctly.

Thank you very much.
Regards.

Liryna, 2015-01-06:
The Debian package need a fix :(

Glad to help you :)

closed 2015-01-06 by Liryna

nimch, 2015-01-06:
For windows issue I will open a new ticket if needed later. For now, I close this one. Too late, done...
Thanks for your help.