liblogicalaccess/liblogicalaccess #75: Link error windows/visual studio 2013

nimch, 2015-01-09:
Hello

I'm trying to build a programme using liblogicalaccess and I can't solve link error.
I have boost 1.57 precompiled installed (lib32-msvc-12.0), openssl 1.0.1j and liblogicalaccess 1.77.1 too.
Compilation is ok but I encounter link error, for exemple : error LNK2019 __declspec(dllimport) public: __thiscall logicalaccess::ReaderConfiguration::ReaderConfiguration(void) in spite of I put lib path in my visual studio solution settings.
Any idea ?

Thanks
Regards

Liryna, 2015-01-09:
Are you sure you have liblogicalaccess.lib as dependency ?

nimch, 2015-01-09:
yes

Liryna, 2015-01-09:
Did you try the basic:
https://github.com/islog/liblogicalaccess/blob/master/samples/basic/
?

nimch, 2015-01-09:
My test programm is more or less the basic but yes I check again : same link error

Liryna, 2015-01-09:
I have really no idea :(
I build my windows project exactly like this.
It seems that your Visual Studio is not looking for the symbole inside the liblogicalaccess.
What happen if you remove liblogicalaccess from the path ?

nimch, 2015-01-09:
With or without liblogicalaccess.lib path give the same link errors.
If I remove lib boost path I have new errors relative to boost so I can conclude this parameter (include) is read...

nimch, 2015-01-09:
Thank you for your help.
Weekend time.
I will investigate next week.
Bye.

Liryna, 2015-01-09:
The project use the same architecture as the library in the path ?

nimch, 2015-01-12:
hello.

Yes, all win32.

Liryna, 2015-01-12:
"With or without liblogicalaccess.lib path give the same link errors."
it mean that Visual Studio is even not using liblogicalaccess.lib.

You should look on this way

Liryna, 2015-01-19:
Have you been able to fix the error ?

nimch, 2015-01-19:
No...

closed 2015-02-04 by Liryna