liblogicalaccess/liblogicalaccess #88: Configuration file for mapping hardcoded ATR to card type

xaqq, 2015-07-01:
Some card types are detected by (fully) comparing the ATR with an hardcoded value.
It would be beneficial to use a configuration file to define this mapping.

Something similar to this:
```
<map>
    <atr>XXXXXXXXXXXXXXXXXXXXXXXXX</atr>
    <cardType>MifarePlus_SL1_4K</cardType>
</map>
<map>
    <atr>XXXXXXXXXXXXXXXXXXXXXXXXZ</atr>
    <cardType>MifarePlus_SL1_2K</cardType>
</map>
```

xaqq, 2015-12-04:
ATR Parsing moved to own class. Will be easier to eventually do that now.
See https://github.com/islog/liblogicalaccess/commit/8734b16f63c0371d389b02454ceb9be682635b7f

Maxhy, 2016-06-11:
ATR parsing class was cleaned on latest commits. Current ATR parsing logic is now acceptable and configuration file is not a requirement anymore (to the flexibility argument: a new LLA version will still be required anyway to provide the auto-detection by default, and it's already possible to force a card type so this config file will only have little benefits).

closed 2016-06-11 by Maxhy