liblogicalaccess/liblogicalaccess #90: C# example code link is out of date and problem with reading Mifare 1K card. Secondly question about 125 Khz reader/writer.

Ogglas, 2015-07-28:
Hello,

First of all I would like to say I really like your library and the effort you seem to have made in making it. However I can't get it to fully work for me, I'm sure there is something that I have missed. I can connect to my reader and it identifies a Mifare 1K card and I can write to it using your example code. I am using a ACR-122 reader/writer.

In the Wiki section for "DotNet (C#)" it says "Download, build and run the C# basic sample.". However there is no link to a sample. I modified the link to the Delphi code and found "basic-csharp" at the URL-below. However this code seems out of date.

http://www.islog-services.eu/software/liblogicalaccess/

I then tried with the code that you provide in the Wiki and it works flawlessly. However when I try to read data I get errors. 

Code below results in the error "Data length too small.". I assume I should use something other than Wiegand37WithFacilityFormat but I cant get it to work.
```C#
var readFormat = acService.ReadFormat(format, mifareLocation, aiToUse);
```

Code below results in the following errors and I don't understand why.
```C#
var readData = storage.ReadData(location, aiToUse, 16, CardBehavior.CB_DEFAULT);
```
"Exception thrown at 0x0F9734CA (liblogicalaccess.com.dll) in RfidLibLogicalAccess.exe: 0xC0000005: Access violation reading location 0x00000000." 

"An unhandled exception of type 'System.AccessViolationException' occurred in RfidLibLogicalAccess.exe
Additional information: Attempted to read or write protected memory. This is often an indication that other memory is corrupt."

-I have tried shutting down processes that I don't need, turning off anti virus etc. Any clue what this can be?

Code below results in the following error:
```C#
var sectors = samcmd.ReadSectors(1, 2, aiToUse.SAB);
```
"An unhandled exception of type 'System.Runtime.InteropServices.COMException' occurred in RfidLibLogicalAccess.exe

Additional information: No key found."

Two methods that I do get data from (16 bytes) are:
```C#
var samcmdSector1 = samcmd.ReadSector(1, aiToUse.SAB, true);
var headerData = storage.ReadDataHeader(location, aiToUse);
```

```C#
                        if (chip is IMifareChip)
                        {
                            MifareLocation location = new MifareLocation();
                            location.Sector = 1;

                            MifareAccessInfo aiToUse = new MifareAccessInfo();
                            MifareAccessInfo aiToWrite = new MifareAccessInfo();
                            aiToWrite.KeyA.Value = "ff ff ff ff ff ff";
                            aiToWrite.KeyB.Value = "";

                            Wiegand37WithFacilityFormat format = new Wiegand37WithFacilityFormat();
                            format.FacilityCode = 100;
                            format.Uid = 1234;

                            StorageCardService storage = chip.GetService(CardServiceType.CST_STORAGE) as StorageCardService;

                            IAccessControlCardService acService = chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;

                            if (acService.WriteFormat(format, location, aiToUse, aiToWrite))
                            {
                                Console.WriteLine("Format wrote successfully.");
                                IMifareCommands samcmd = chip.Commands as IMifareCommands;
                                samcmd.Authenticate(location, aiToUse);

                                var samcmdSector1 = samcmd.ReadSector(1, aiToUse.SAB, true);
                                var headerData = storage.ReadDataHeader(location, aiToUse);

                                //var sectors = samcmd.ReadSectors(1, 2, aiToUse.SAB);
                                //var readFormat = acService.ReadFormat(format, location, aiToUse);
                                //var readData = storage.ReadData(location, aiToUse, 16, CardBehavior.CB_DEFAULT);
                            }
                            else
                            {
                                Console.WriteLine("Error: Cannot write the format.");
                            }
                        }
```

Secondly I would like to ask about the support for 125 Khz reader/writer. Do you think the reader/writer below with a ATA5577 chip would work? If not which reader/writer would you recommend for 125 Khz?

http://www.ebay.com/itm/USB-125Khz-RFID-EM4305-T5567-Card-Reader-Writer-Copier-Writer-programmer-burner-/371289466277?pt=LH_DefaultDomain_0&hash=item567293a5a5

Liryna, 2015-07-28:
Hello,

You are right! the link for the C# sample has been removed because it was unmaintained. I will also remove all the sentence. Wiki examples should be enough and more easy to keep update. 

For Delphi code, we know that we are compatible because of the COM library but unfortunately none of the team are using it so it is  impossible for us to maintain it.

```
var readData = storage.ReadData(location, aiToUse, 16, CardBehavior.CB_DEFAULT);
```
Looks like you have a c++ segv but debug it from c# is pretty hard :smile:. Could you provide logs ?
https://github.com/islog/liblogicalaccess/wiki/Frequent-issues-and-questions#how-to-activate-some-traces

About your read issue, could you try this:
```
MifareKey key = new MifareKey();
key.Value = "ff ff ff ff ff ff";
GetCommands().LoadKeyNo(0, key, MifareKeyType.KT_KEY_A);
GetCommands().AuthenticateKeyNo(1, 0, MifareKeyType.KT_KEY_A);
GetCommands().ReadBinary(1, 16);

SectorAccessBits sab = new SectorAccessBits();
object data = GetCommands().ReadSector(2, sab, false);
```
or
```
IAccessControlCardService acService = chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;
Assert.IsNotNull(acService);

IMifareLocation location = new MifareLocation();
location.Sector = 2;
IWiegand26Format format = new Wiegand26Format();
format.Uid = 1000;
format.FacilityCode = 67;
Assert.IsTrue(acService.WriteFormat(format, location, null, null));

IAccessControlCardService acService = chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;
Assert.IsNotNull(acService);

IMifareLocation location = new MifareLocation();
location.Sector = 2;
IWiegand26Format format = new Wiegand26Format();
IWiegand26Format rformat = acService.ReadFormat(format, location, null) as IWiegand26Format;
Assert.IsNotNull(rformat);
Assert.IsTrue(rformat.Uid == 1000 && rformat.FacilityCode == 67, "Bad format data.");
```

About 125Khz reader,
We are compatible with different 125Khz reader BUT none of them are able to write.
ATA5577 chip are unfamiliar for me so I would say no but if you could test, it would be nice !

We have the reader RFID EM4305 T5567 but had not time and need to implement it. It is not a pcsc reader so it need a special implementation but it should be simple to do it.

Ogglas, 2015-07-29:
Thanks for your quick reply! I'm having trouble getting logging to work. I downloaded the library via NuGet and have placed liblogicalaccess.config next to Interop.LibLogicalAccess.Com.dll. Do I need to edit "/usr/lib/logicalaccess" or $current?

```XML
<?xml version="1.0" encoding="utf-8"?>
<config>
	<log>
		<enabled>true</enabled>
		<filename>liblogicalaccess.log</filename>
		<to_stderr>false</to_stderr>
		<seewaitinsertion>true</seewaitinsertion>
		<seewaitremoval>true</seewaitremoval>
		<seecommunication>true</seecommunication>
		<seeplugin>true</seeplugin>
	</log>
	<autodetect>
		<enabled>true</enabled>
		<timeout>400</timeout>
	</autodetect>
	<retrySerialConfiguration>
		<enabled>true</enabled>
		<timeout>500</timeout>
	</retrySerialConfiguration>
	<reader>
		<default>PCSC</default>
	</reader>
	<PluginFolders>
		<Folder>$current</Folder>
		<Folder>/usr/lib/logicalaccess</Folder>
	</PluginFolders>
</config>
```

Unfortunately the code examples you supplied does not work.
```C#
IMifareCommands samcmd = chip.Commands as IMifareCommands;

MifareKey key = new MifareKey();
key.Value = "ff ff ff ff ff ff";
samcmd.LoadKeyNo(0, key, MifareKeyType.KT_KEY_A);
samcmd.AuthenticateKeyNo(1, 0, MifareKeyType.KT_KEY_A);
samcmd.ReadBinary(1, 16);

SectorAccessBits sab = new SectorAccessBits();
object data = samcmd.ReadSector(2, sab, false);
```
Returns byte[0] and if I change sector to 1 like this:
```C#
object data = samcmd.ReadSector(1, sab, false);
```
I get the No key found exception. "An unhandled exception of type 'System.Runtime.InteropServices.COMException' occurred in RfidLibLogicalAccess.exe
Additional information: No key found."

In the other example the commented out code below works but the second still generates "Data length too small". Test method TestRfid.UnitTest1.TestMethod1 threw exception: 
System.Runtime.InteropServices.COMException: Data length too small.

```C#
var chip = readerUnit.GetSingleChip();
//IAccessControlCardService acService =
//    chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;
//Assert.IsNotNull(acService);

//IMifareLocation location = new MifareLocation();
//location.Sector = 2;
//IWiegand26Format format = new Wiegand26Format();
//format.Uid = 1000;
//format.FacilityCode = 67;
//Assert.IsTrue(acService.WriteFormat(format, location, null, null));

IAccessControlCardService acService = chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;
Assert.IsNotNull(acService);

IMifareLocation location = new MifareLocation();
location.Sector = 2;
IWiegand26Format format = new Wiegand26Format();
IWiegand26Format rformat = acService.ReadFormat(format, location, null) as IWiegand26Format;
Assert.IsNotNull(rformat);
Assert.IsTrue(rformat.Uid == 1000 && rformat.FacilityCode == 67, "Bad format data.");
```

Ok I understand. Do you have any plans to enable write for 125Khz? I have bought a reader / writer using ATA5577 and I will let you know. They only cost around 20$ if you want a piece that can write as well. :)

Liryna, 2015-07-29:
liblogicalaccess.config should be placed side to side with liblogicalaccess.dll.
Normally the windows installer put it in the good place.

I found the issue, my example was wrong ReadSector need the keys to be set in the Mifareprofile.
```
IMifareCommands cmd = chip.Commands as IMifareCommands;

MifareKey key = new MifareKey();
key.Value = "ff ff ff ff ff ff";
cmd.LoadKeyNo(1, key, MifareKeyType.KT_KEY_A);
cmd.AuthenticateKeyNo(1, 0, MifareKeyType.KT_KEY_A);
object data = cmd.ReadBinary(1, 16);

SectorAccessBits sab = new SectorAccessBits();
(chip.Profile as IMifareProfile).SetKey(1, MifareKeyType.KT_KEY_A, "ff ff ff ff ff ff");
data = cmd.ReadSector(1, sab, false);
```

Second case that made it work for me:
Write:
```
IAccessControlCardService acService =
chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;

MifareAccessInfo aiToUse = new MifareAccessInfo();
MifareAccessInfo aiToWrite = new MifareAccessInfo();
aiToWrite.KeyA.Value = "ff ff ff ff ff ff";
aiToWrite.KeyB.Value = "";

IMifareLocation location = new MifareLocation();
location.Sector = 2;
IWiegand26Format format = new Wiegand26Format();
format.Uid = 1000;
format.FacilityCode = 67;
acService.WriteFormat(format, location, aiToUse, aiToWrite);
```
Read:
```
IAccessControlCardService acService = chip.GetService(CardServiceType.CST_ACCESS_CONTROL) as IAccessControlCardService;

IMifareLocation location = new MifareLocation();
location.Sector = 2;
IWiegand26Format format = new Wiegand26Format();
IWiegand26Format rformat = acService.ReadFormat(format, location, null) as IWiegand26Format;
```
Write for 125Khz is not planned for now :cry: but If you feel to want to contribute, I would be happy to help you ! since we already have the reader, it gonna even be easy to test.

Ogglas, 2015-07-30:
Thanks Lyrina, I can now read all 16 sectors. Will add the code here if it can help someone else. 

```C#
IMifareCommands samcmd = chip.Commands as IMifareCommands;

MifareKey key = new MifareKey();
key.Value = "ff ff ff ff ff ff";
samcmd.LoadKeyNo(1, key, MifareKeyType.KT_KEY_A);
samcmd.AuthenticateKeyNo(1, 0, MifareKeyType.KT_KEY_A);
samcmd.ReadBinary(1, 16);

SectorAccessBits sab = new SectorAccessBits();

for (int i = 0; i < 16; i++)
{
    (chip.Profile as IMifareProfile).SetKey(i, MifareKeyType.KT_KEY_A, "ff ff ff ff ff ff");
}

var list = new List<List<char>>();

for (int i = 0; i < 16; i++)
{
    try
    {
        object data = samcmd.ReadSector(i, sab, true);
        IEnumerable enumerable = data as IEnumerable;
        if (enumerable != null)
        {
            var listChar = new List<char>();

            foreach (object element in enumerable)
            {
                int intValue = Convert.ToInt32(element.ToString(), 10);
                char c = Convert.ToChar(intValue);
                listChar.Add(c);
            }
            list.Add(listChar);
            var completeString = string.Concat(list);
            //var utf8String = Encoding.UTF8.GetString(byteArray);
        }
    }
    catch (Exception ex)
    {
        if (ex.Message == "No key found.")
        {
            Debug.WriteLine("No key found for sector: " + i);
        }
        else
        {
            throw;
        }
    }
}
```

Liryna, 2015-07-30:
Example added to the wiki :)
https://github.com/islog/liblogicalaccess/wiki/Use-this-RFID-library-in-DotNet-%28C%23%29#read-all-16-sectors-of-a-mifare

Have you seen some misleading example on this page ?

Ogglas, 2015-08-03:
Thank you Liryna! 

No the examples you have on the page work, at least for Mifare Classic 1K.

Liryna, 2015-08-03:
Great ! I close the issue so :+1:

closed 2015-08-03 by Liryna