liblogicalaccess/liblogicalaccess #93: Quick question about Desfire Ev2

clarklight, 2015-10-09:
I saw a page that says "The chip Mifare DESFire EV2 is not released yet by NXP", and that was edited in June 2015, but the first document about releasing EV2 was in 2013...i am just wondering it is officially out and can be purchased on the market??

Liryna, 2015-10-09:
It is not officially out for now :( We are still waiting

clarklight, 2015-10-09:
Thanks.... yer it just seems like its been a while...i got a quick question not sure if you have any experience on it.... with NTAG 21x family, their password protected area, do you think its secure? I read on their documents and some opinion that it is pretty weak. 

I am just trying to figure out for "Anti-cloning" mechanism if NTAG 21x or Desfire would be better.

Liryna, 2015-10-09:
Well NTAG 21x are not made for security. If someone want to clone it and with a little experience, he can do it.
But as always, it depend of the security you need.

clarklight, 2015-10-09:
Em.... yer, thats what i was thinking.... i just dont want any card to be cloned. Was planning to have a password protected area that has an account ID hidden within it. and Only when password has been entered correctly then it can be read.

So i guess i will have to go with the Desfire EV1 for now. Thanks for the reply!

Liryna, 2015-10-09:
No problem :) If you need any other help, feel free to ask !

closed 2015-10-09 by Liryna