liblogicalaccess/liblogicalaccess #94: NXP Ultralight EV1 security same as NTAG 21x 32bytes password? Or Desfire 3DES?

clarklight, 2015-10-10:
I got a quick question regarding the security for Ultralight EV1, just wondering if you can help me out on it. 

I read the Ultralight has a 32byte password protection, wondering if thats the same as the NTAG21x password layer, or if thats same security level as the Desfire(3DES/AES)etc? 

My use case is that i dont need much memory space just 200-300 bytes or so, but i need higher level of security. Wondering if you would know about the Ultralight EV1, thanks!

Liryna, 2015-10-10:
Yes, it seems that Ultralight EV1 authentication is the same logic as NTAG21x.
The authentication by password is not secure because it is transmit without protection. 
This card have no mechanism of secure session & channel.

It mean that someone listen your communication can get your password, the data that you read and even can send commands if he want.

DESFireEV1 has this mechanism.

clarklight, 2015-10-12:
Sweet, thanks for the reply!!

closed 2015-10-12 by Liryna