liblogicalaccess/liblogicalaccess #96: "not enough storage" exception if no reader driver

AndrewLeiper, 2015-10-21:
Hi there,

I'm using liblogicalaccess-exe-1.79.1 from delphi.

If no reader driver is installed, the following line generates an EOleSysError exception with the message "not enough storage is available to complete this operation".

readerProvider := CoPCSCReaderProvider.Create;

After I install the omnikey driver, it works fine and in fact it continues to work fine even if I uninstall the driver again. (ie it produces a more helpful "No reader found" message).

My guess would be that installing the driver activates some PCSC functionality in windows and until that happens the library doesn't trap the error resulting in the misleading "Not enough storage" exception from the com dll.

This happened on several windows 7 and 10 PCs. Not a big problem when you know what's going on obviously.

Thanks,

Andy Leiper

Liryna, 2015-10-21:
Hi @AndrewLeiper ,

You are right, the driver does enable some PCSC functionality in windows.
I would guess that he enabled the pcsc service. On some windows, this service is disabled by default.
So the PCSC initialisation calls does not work and we throw "not enough storage is available to complete this operation".

AndrewLeiper, 2015-10-22:
Hi Liryna,

Thanks. You are absolutely right. If I stop the smartcard service, that causes the same error.

Andy Leiper

Liryna, 2015-10-22:
Great :+1: that it helped you !

closed 2015-10-22 by Liryna