liblogicalaccess/liblogicalaccess #98: COMException: Wrong Instruction

EniacMlezi, 2015-11-10:
Hey everybody,

I'm getting COMException was unhandled error. I've tried every possible option to either read or write, but the error returns every time. I'm using SCL3711 reader and try to read a mifare 1k card. 
[image]

Liryna, 2015-11-10:
Hi,

Could provide some logs about the crash ?
https://github.com/islog/liblogicalaccess/wiki/Frequent-issues-and-questions#how-to-activate-some-traces

EniacMlezi, 2015-11-10:
Thanks for the fast response!

https://onedrive.live.com/redir?resid=A38BBFADE71C597D!1201&authkey=!AB7TfEez4vcBnGs&ithint=folder%2c

here's a link to an activityLog and error discription

Liryna, 2015-11-10:
I wanted to logicalaccess.log but the exception message seems to be enouth :smile: 

You have the error: PCSC error: 80100069. The smart card has been removed, so further communication is not possible.

Are you sure the card does not move or something like ?
I don't think it is a liblogicalaccess issue. the error come from pcsc directly :( he tell us that he lost the communication with the contactless card.

EniacMlezi, 2015-11-10:
Saw it just now.. it's kinda sad, because i'm definitely not moving the card. I may be able to borrow a reader from a teacher. If that reader is compatible, could it possibly fix the problem? And what about testing it on cpp instead?

Liryna, 2015-11-10:
I just made the test again with our software and the SCL3711 reader work perfectly.
Are you using a NFCReaderUnit ?
and Zadigt to use the libUSB0 v1.2.6.0 driver http://zadig.akeo.ie/ ?

EniacMlezi, 2015-11-10:
No, I was using IReaderUnit.
Just installed zadig and tried using NFCReaderUnit, i'm now getting this error:

[image]
[image]

full program and error discription:
https://onedrive.live.com/redir?resid=A38BBFADE71C597D!1204&authkey=!AItKhtfYibmOKq0&ithint=folder%2c

Liryna, 2015-11-10:
Could you try with readcard: http://www.islog.com/en/software/readcard/
If it does not work, it mean there is an issue with the driver.
Otherwise it is with the code.

EniacMlezi, 2015-11-10:
readcard succeeds, 
[image]

I cracked the keys for sector 10-14 with mfoc. It was really unstable and often lost connection as well. After i ran pcscd -f (linux, sort of daemon tool for pcsc) i managed to keep a connection open for long enough to crack it. This may be a similar problem?

Liryna, 2015-11-10:
Possible but our reader work perfectly :( I do not know what is the difference

xaqq, 2015-11-10:
Hello,

It seems the SCL3711 can be used either as a PCSC reader (through
liblogicalaccess' PCSC Provider) or as
a NFC reader using the liblogicalaccess' NFC provider.
To use it with the NFC Provider you need to use the libusb driver.
Otherwise it defaults to behave like a PCSC Reader.

So, I guess we're back to square one. When using a PCSC Provider you
somehow lose connectivity with the smart card.
Did you experience any similar bug with readcard (loss of connectivy with
card) ?

If anything I'd try with the C++ API directly. Less layers means less
potentials bugs and more chance to diagnose the problem.

EniacMlezi, 2015-11-10:
nope, no problems with readcard at all.. i'm not that big a c++ programmer but I'll take a look at it tonight.

Have you already tested my application to verify it's not in my code?
https://onedrive.live.com/redir?resid=A38BBFADE71C597D!1204&authkey=!AItKhtfYibmOKq0&ithint=folder%2c

Liryna, 2015-11-10:
No longer at work :cry:

EniacMlezi, 2015-11-10:
naah no problem, i've got 10 more weeks to deliver this school project.

EniacMlezi, 2015-11-10:
SO i'm currently stuck on simply including logicalaccess in cpp, i'm really not familiar... Do any of you have a solution lying around?

Liryna, 2015-11-10:
@EniacMlezi You should start from the basic. Look at the command of Travis if you have issue to build it.

EniacMlezi, 2015-11-11:
Heey, 
still no progress on the c# lib. I've (finally) succesfully built the logicalaccess library, which was a real pain in the ass since I haven't done anything like it before... I tried compiling some code but got these errormessages: 
-------------- Build: Debug in RFID_test (compiler: GNU GCC Compiler)---------------

g++ -Wall -fexceptions -g -std=c++11  -c /home/lars/CodeBlocks/RFID_test/main.cpp -o obj/Debug/main.o
In file included from /usr/local/include/logicalaccess/dynlibrary/librarymanager.hpp:7:0,
                 from /home/lars/CodeBlocks/RFID_test/main.cpp:11:
/usr/local/include/logicalaccess/dynlibrary/idynlibrary.hpp:23:32: error: expected ‘)’ before ‘*’ token
     typedef void (CDECL_WIN32_ *getprovider)(std::shared_ptr<ReaderProvider>*);
                                ^
/usr/local/include/logicalaccess/dynlibrary/idynlibrary.hpp:24:32: error: expected ‘)’ before ‘*’ token
     typedef void (CDECL_WIN32_ *getcard)(std::shared_ptr<Chip>*);
                                ^
/usr/local/include/logicalaccess/dynlibrary/idynlibrary.hpp:25:32: error: expected ‘)’ before ‘*’ token
     typedef void (CDECL_WIN32_ *getdiversification)(std::shared_ptr<KeyDiversification>*);
                                ^
/usr/local/include/logicalaccess/dynlibrary/idynlibrary.hpp:26:32: error: expected ‘)’ before ‘*’ token
     typedef void (CDECL_WIN32_ *getcommands)(std::shared_ptr<Commands>*);
                                ^
/usr/local/include/logicalaccess/dynlibrary/idynlibrary.hpp:27:32: error: expected ‘)’ before ‘*’ token
     typedef void (CDECL_WIN32_ *setcryptocontext)(std::shared_ptr<Commands>*, std::shared_ptr<Chip>*);
                                ^
/usr/local/include/logicalaccess/dynlibrary/idynlibrary.hpp:28:32: error: expected ‘)’ before ‘*’ token
     typedef bool (CDECL_WIN32_ *getobjectinfoat)(unsigned int, char*, size_t, void**);
                                ^
In file included from /home/lars/CodeBlocks/RFID_test/main.cpp:11:0:
/usr/local/include/logicalaccess/dynlibrary/librarymanager.hpp:23:36: error: expected initializer before ‘LibraryManager’
     class LIBLOGICALACCESS_API_DLL LibraryManager
                                    ^
/home/lars/CodeBlocks/RFID_test/main.cpp:30:1: error: expected ‘}’ at end of input
 }
 ^
Process terminated with status 1 (0 minute(s), 1 second(s))
8 error(s), 0 warning(s) (0 minute(s), 1 second(s))
 
I've tried eclipse, codeblocks and commandline compiling... nothing seems to work for me

EniacMlezi, 2015-11-11:
https://onedrive.live.com/redir?resid=A38BBFADE71C597D!1205&authkey=!AJGFUCgstuvD4gw&ithint=file%2c

here's a link to the file

Liryna, 2015-11-11:
Add `-DUNIX` to your command line

EniacMlezi, 2015-11-11:
greaat thanks! I now only get some link errors; "undefined reference to: ". I include files from /usr/local/logicalaccess

Liryna, 2015-11-12:
Yes you have to link with logicalaccess and all module that you are using like:
-L/usr/local/logicalaccess -llogicalaccess -ldesfirecards  ....

EniacMlezi, 2015-11-12:
man this c++ stuff is complicated.. haha. Still haven't been able to run a single binary.. when trying to install the library via 'sudo apt-get install liblogicallaccess' i get libboost-date-time dependency error. And when i try to install boost via apt-get it gives even more dependeny errors... it's really driving me mad by now hehe.

I' ve tried to build them both manually, but i'm not too sure if that worked out. After building boost i can see its path /usr/local/include/boost, and after building liblogicalaccess i've copied the include folder to /usr/local/include. That folder only contains header files, so i'd have to use the above technique to link them, right?

And besides, I'm getting an error with std::shared_ptr, symbol could not be resolved in eclipse. I believe it has something to do with eclipse screwing up c++11. Saw someone who said it would be fixed by adding -std=c++11. And it did fix it when compiling via command line, but not in eclipse. Perhaps you know more? 

I really appreciate your help, learned more about linux repo's in the past few days than i've ever learned about it before ;)

Liryna, 2015-11-12:
Ok ok :smile: 
Lets start from a clean base.
```
 apt-get remove liblogicallaccess
 apt-get install g++ gcc cmake
 apt-get install libboost1.55-dev libboost-system1.55-dev libboost-filesystem1.55-dev libboost-date-time1.55-dev libboost-chrono1.55-dev libboost-thread1.55-dev libboost-regex1.55-dev
 apt-get install libpcsclite-dev libssl-dev
 git clone https://github.com/islog/liblogicalaccess.git
 cd liblogicalaccess
 mkdir build && pushd build && cmake -DCMAKE_INSTALL_PREFIX=$PWD/install .. && make install && popd && cd samples/basic/ && mkdir build && pushd build && cmake -DLibLogicalAccess_DIR=$PWD/../../../build/install/lib/cmake/liblogicalaccess .. && make
```

Tell me when all of this work.

EniacMlezi, 2015-11-12:
Haha a clean base, currently installing a new VM

EniacMlezi, 2015-11-12:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libboost-chrono1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libboost-date-time1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libboost-filesystem1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libboost-regex1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libboost-system1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libboost-thread1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libboost1.55-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libboost1.55-dev' has no installation candidate
E: Package 'libboost-system1.55-dev' has no installation candidate
E: Package 'libboost-filesystem1.55-dev' has no installation candidate
E: Package 'libboost-date-time1.55-dev' has no installation candidate
E: Package 'libboost-chrono1.55-dev' has no installation candidate
E: Package 'libboost-thread1.55-dev' has no installation candidate
E: Package 'libboost-regex1.55-dev' has no installation candidate

These occur. after this i tried the boost-latest ppa and did the above, but got 404 errors. So i believe the only other option was building it from its source.

Liryna, 2015-11-12:
You have probably a newer version of boost than 1.55. Try:

`apt-get install libboost-dev libboost-system-dev libboost-filesystem-dev libboost-date-time-dev libboost-chrono-dev libboost-thread-dev libboost-regex-dev`

EniacMlezi, 2015-11-12:
okay got it, all packages installed; boost 1.58 and the lib is built. Do you recommend installing an Ide like eclipse, or should i just use a text editor?

Liryna, 2015-11-12:
As you want for this part :) but you will use command line to build so eclipse is only going to be here to edit files.
This command line: `mkdir build && pushd build && cmake -DCMAKE_INSTALL_PREFIX=$PWD/install .. && make install && popd && cd samples/basic/ && mkdir build && pushd build && cmake -DLibLogicalAccess_DIR=$PWD/../../../build/install/lib/cmake/liblogicalaccess .. && make`
build two main projects. The lib and a basic in samples/basic

Copy all library and https://github.com/islog/liblogicalaccess/blob/master/liblogicalaccess.config in the samples/basic folder.
You should be able to run it.

After edit the basic.cpp, add your code little by little.
To build: `make` samples/basic

xaqq, 2015-11-12:
The IDE doesn't matter much, it's a matter of personal preference. Use what
you're confortable with.

Note that the library uses C++11 so if the IDE doesn't support it properly
this may be annoying (or confusing, especially if you're new to C++).

EniacMlezi, 2015-11-12:
@xaqq yeah saw it, found a fix for eclipse. But i'll just use a text editor then :+1:

EniacMlezi, 2015-11-12:
@Liryna what exactly do you mean by 'copy all library' ?

Liryna, 2015-11-12:
in $PWD/install/lib you have the dynamic library (.so) that need to be near the basic.

EniacMlezi, 2015-11-12:
I can finally run c++ code :) next problem: the reader is not detected on the ubuntu vm, already installed libusb and tried libnfc but no result

Liryna, 2015-11-12:
Great :trophy: !
Did you install the drivers (I never did in linux so I will not be able to help you for this part)

xaqq, 2015-11-12:
IIRC you don't need any driver in particular.
What's the output of `lsusb`?

Maybe you simply forgot to configure USB passthrough for the reader
(telling your hypervisor to give control to the USB device to the VM).

EniacMlezi, 2015-11-12:
well i'm using virtualbox to run ubuntu, and you were right; virtualbox doesn't passthrough any higher version than usb 1.1 without some extensions. But after installing them there's still no change. 

lsusb returns this:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

EniacMlezi, 2015-11-12:
Ow, found something. in virtualbox one also needs to specify the usb device to be passed:
[link](https://www.howtoforge.com/virtualbox-2-how-to-pass-through-usb-devices-to-guests-on-an-ubuntu-8.10-host)

EniacMlezi, 2015-11-13:
goodmorning! last night i got the reader working on ubuntu. I ran into some problems with automatically loaded modules -> [fix](http://forums.nfc-tools.org/topic/668/solved-scl3711-interface-0-claimed-by-pn533-nfclist-sets-conf/)

Now when i run the basic example it only says:
Available reafer plug-ins (0):
Available card plug-ins (0):

and then it terminates... 

With some ghetto debugging(no idea how to debug this stuff) i founf out the bug is in the if statement: 

<pre><code>std::shared_ptr<logicalaccess::ReaderConfiguration> readerConfig(new logicalaccess::ReaderConfiguration());
std::cout << "DEBUG" << std::endl;
// PC/SC
   readerConfig->setReaderProvider(logicalaccess::LibraryManager::getInstance()->getReaderProvider("PCSC"));
std::cout << "DEBUG0" << std::endl;
if (readerConfig->getReaderProvider()->getRPType() == "PCSC" && readerConfig->getReaderProvider()->getReaderList().size() == 0)
{
            std::cerr << "No readers on this system." << std::endl;
            return EXIT_FAILURE;
}
std::cout << "DEBUG1" << std::endl;
std::cout << readerConfig->getReaderProvider()->getReaderList().size() << " readers on this system." << std::endl;
std::cout << "DEBUG2" << std::endl;</pre></code>

DEBUD and DEBUG0 are printed, but "No readers on this system." isn't. Any ideas?

EniacMlezi, 2015-11-13:
Neither are DEBUG1 and DEBUG2 by the way, it just seems to terminate before or after that loop

Liryna, 2015-11-13:
When you have:
```
Available reafer plug-ins (0):
Available card plug-ins (0):
```

It mean that it have not found the library like pcscreaders.so, desfirecards.so...
Did you copy well all .so in the same folder as the basic with the liblogicalaccess.config ?

The config file is a xml where you can specify the PluginFolders ( where you have the .so).

EniacMlezi, 2015-11-13:
just copied the full contents of liblogicalaccess/build/install/lib to sampes/basic/build(thought only liblogicalaccess.so and .config mattered, i know bit stubborn), now it gives 15 reader plug-ins, 53 card plug-ins and "no readers on this system". Libnfc can detect the reader, so that's not the problem

Liryna, 2015-11-13:
Ok great !
So the reader is seen as nfc and not pcsc on linux.

You have to build the NFC plugin here: https://github.com/islog/liblogicalaccess-libnfc
clone it and try: `mkdir build && pushd build && cmake -DCMAKE_INSTALL_PREFIX=$PWD/install .. && make install`

Copy the new .so build, in this repository, with your basic also.
Change "PCSC" in your code by "NFC".

EniacMlezi, 2015-11-13:
building it his way i get
fatal error: logicalaccess/readerproviders/readerunitconfiguration.hpp: no such file or directory

i am trying to build it from my home dir, and i've previously built liblogicalaccess in my home dir as well. in order for the libnfc-plugin compilation to find these missing files, should i define some sort of path to the full liblogicalaccess in the cmake command or built it someplace else(like in /home/lars/liblogicalaccess)?

Liryna, 2015-11-13:
Yes sorry I forgot: You have to set LIBLOGICALACCESS_INCLUDE_DIR & LIBLOGICALACCESS_PLUGINS_DIR
Something like:
`mkdir build && pushd build
cmake -DCMAKE_INSTALL_PREFIX=$PWD/install -DLIBLOGICALACCESS_INCLUDE_DIR=/home/user/liblogicalaccess/include -DLIBLOGICALACCESS_PLUGINS_DIR=/home/user/liblogicalaccess/build/install/lib  .. && make install`

EniacMlezi, 2015-11-13:
it now gets past the previous fatal error, but there's a new one, this time it can't find:

readercardadapters/iso7816readercardadapter.hpp

Liryna, 2015-11-13:
it is when you build the liblogicalaccess-libnfc ?
Can you give the full error output

EniacMlezi, 2015-11-13:
~/liblogicalaccess-libnfc/build$ cmake -DCMAKE_INSTALL_PREFIX=$PWD/install -DLIBLOGICALACCESS_INCLUDE_DIR=/home/lars/liblogicalaccess/include -DLOGICALACCESS_PLUGINS_DIR=/home/lars/liblogicalaccess/build/install/lib .. && make install
-- Try C++11 flag = [-std=c++11]
-- Performing Test CXX11_FLAG_DETECTED
-- Performing Test CXX11_FLAG_DETECTED - Success
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   date_time
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lars/liblogicalaccess-libnfc/build
Scanning dependencies of target libnfc-nfcreaders
[ 12%] Building CXX object plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/nfcreaderunitconfiguration.cpp.o
[ 25%] Building CXX object plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/libraryentry.cpp.o
[ 37%] Building CXX object plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/nfcdatatransport.cpp.o
[ 50%] Building CXX object plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/nfcreaderprovider.cpp.o
[ 62%] Building CXX object plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/nfcreaderunit.cpp.o
In file included from /home/lars/liblogicalaccess-libnfc/plugins/pluginsreaderproviders/nfc/nfcreaderunit.cpp:26:0:
/home/lars/liblogicalaccess-libnfc/plugins/pluginsreaderproviders/nfc/readercardadapters/nfcreadercardadapter.hpp:10:59: fatal error: readercardadapters/iso7816readercardadapter.hpp: No such file or directory
compilation terminated.
plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/build.make:146: recipe for target 'plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/nfcreaderunit.cpp.o' failed
make[2]: *** [plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/nfcreaderunit.cpp.o] Error 1
CMakeFiles/Makefile2:90: recipe for target 'plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/all' failed
make[1]: *** [plugins/pluginsreaderproviders/nfc/CMakeFiles/libnfc-nfcreaders.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

Liryna, 2015-11-13:
Do you have this file ?
/home/lars/liblogicalaccess/include//logicalaccess/plugins/cards/iso7816/readercardadapters/iso7816readercardadapter.hpp

EniacMlezi, 2015-11-13:
no, i do have it in 
/home/lars/liblogicalaccess/plugins/pluginsreaderproviders/iso7816/readercardadapters.hpp

Liryna, 2015-11-13:
oh, thats not normal O_o.
Copy it in /home/lars/liblogicalaccess/include//logicalaccess/plugins/cards/iso7816/readercardadapters

EniacMlezi, 2015-11-13:
after /home/lars/liblogicalaccess/include//logicalaccess/ i don't even have a plugins folder... 

in the liblogicalaccess/build/plugins folder there is an sio7816 directory that contains libiso7816readers.so, and i believe its headers are located in liblogicalaccess/build/install/include/logicalaccess/plugins/readers/iso7816/readercardadapters.hpp

maybe we're using the wrong headerfiles? -DLIBLOGICALACCESS_INCLUDE_DIR=/home/user/liblogicalaccess/include doesn't point to the compiled libraries in the build directory

Liryna, 2015-11-13:
Ahhhhhhhhhh sorry I have not seen ><.
Yes change it to /home/lars/liblogicalaccess/build/install/include

EniacMlezi, 2015-11-13:
yeah just realised myself haha, next: 
/usr/bin/ld cannot find -*list of cards -> ldesfirecards, lmifarecards, etc* and -liso7816readers -llogicalaccess

That's probably the linking issue as described a few messages above. I should do it like this for every single one:
-L/usr/local/logicalaccess -llogicalaccess -ldesfirecards ...
right?

Shouldn't this have happened automatically when liblogicalaccess was built?

Liryna, 2015-11-13:
Normally he should find it in the folder you specified:
-DLOGICALACCESS_PLUGINS_DIR=/home/lars/liblogicalaccess/build/install/lib

EniacMlezi, 2015-11-13:
it gives an error for -ldesfirecards, but /build/install/lib contains files like libdesfirecards. I recon there might be a slight problem here...

EniacMlezi, 2015-11-13:
shouldn't -DLOGICALACCESS_PLUGINS_DIR be -DLIBLOGICALACCESS_PLUGINS_DIR

Liryna, 2015-11-13:
Yes, you have copy paste it wrongly :p

EniacMlezi, 2015-11-13:
changed it, now it says :

<pre><code>
cmake -DCMAKE_INSTALL_PREFIX=$PWD/install -DLIBLOGICALACCESS_INCLUDE_DIR=/home/lars/liblogicalaccess/build/install/include -DLIBLOGICALACCESS_PLUGINS_DIR=/home/lars/liblogicalaccess/build/install/lib .. && make install
-- Try C++11 flag = [-std=c++11]
-- Performing Test CXX11_FLAG_DETECTED
-- Performing Test CXX11_FLAG_DETECTED - Success
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   date_time
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lars/liblogicalaccess-libnfc/build
Linking CXX shared library liblibnfc-nfcreaders.so
[ 87%] Built target libnfc-nfcreaders
Scanning dependencies of target mifare1k_test_change_uid
[100%] Building CXX object tests/mifare/CMakeFiles/mifare1k_test_change_uid.dir/test_change_uid.cpp.o
In file included from /home/lars/liblogicalaccess-libnfc/tests/mifare/test_change_uid.cpp:9:0:
/home/lars/liblogicalaccess-libnfc/plugins/pluginsreaderproviders/nfc/nfcreaderunit.hpp:10:56: fatal error: logicalaccess/readerproviders/readerunit.hpp: No such file or directory
compilation terminated.
tests/mifare/CMakeFiles/mifare1k_test_change_uid.dir/build.make:54: recipe for target 'tests/mifare/CMakeFiles/mifare1k_test_change_uid.dir/test_change_uid.cpp.o' failed
make[2]: *** [tests/mifare/CMakeFiles/mifare1k_test_change_uid.dir/test_change_uid.cpp.o] Error 1
CMakeFiles/Makefile2:155: recipe for target 'tests/mifare/CMakeFiles/mifare1k_test_change_uid.dir/all' failed
make[1]: *** [tests/mifare/CMakeFiles/mifare1k_test_change_uid.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
</pre></code>

It does say it's at 100% so we're almost there.... i hope

Liryna, 2015-11-13:
Comment this line:
https://github.com/islog/liblogicalaccess-libnfc/blob/master/CMakeLists.txt#L93

EniacMlezi, 2015-11-13:
Okay great! it's finally built, the reader is detected and card insertion works, werk back at the beginning haha. i've only got some more errors on the reader being occupied by pn553 when a card is removed, but let's just blame it on the vm for now... I'll post if i can actually read from a card(the reason for this thread in the first place, if you can remember haha) this afternoon.

Anyways, thanks for the great help so far!

Liryna, 2015-11-13:
a pleasure :+1: Feel free to recommend the library at your school :yum:

[6 more replies]

closed 2015-11-13 by Liryna