liblogicalaccess/liblogicalaccess #103: Mifare 1K how to add and read values to card

SoftWorksCEO, 2016-04-29:
Dear community,

I just got my first RFID / NFC-reader and writer and was able to retrieve informations like the name of the reader, the card type (Mifare 1K in my case) as well as the UID of the found card in C# on the .NET.platform.

For my scenario I just would like to write a name as well as a GUID to the card - but I've spent an entire day now trying to find something in the docs - but to no avail.

So I would kindly ask you, if you could profice the community with a working sample für c# that shows how to add string values (like a name or a GUID) to the card and how to read these value(s)? That would help so much.

Looking forward to hearing from you,

best regards,
Marco

Liryna, 2016-04-29:
Hi @SoftWorksCEO ,

Have you seen the Mifare read exemple ?
https://github.com/islog/liblogicalaccess/wiki/Use-this-RFID-library-in-DotNet-%28C%23%29#read-all-16-sectors-of-a-mifare

You can use WriteBinary for writing your string in byte[].

SoftWorksCEO, 2016-04-29:
Hey Liryna,

yes, I saw that but since that is for DESFire which I don't have (MIfare 1K Classic - I would guess), that doesn't help me.

I am completely new to that topic - so could you hand me some code for that? I could immediately try it.

Best regards,
Marco

SoftWorksCEO, 2016-04-29:
In the sample, you provide, the compiler says: "location must be a MifareLocation."

This is why I can't use it :-(

Liryna, 2016-04-29:
>Read All 16 sectors of a Mifare

This part of the page is for MIfare 1K Classic

```
IMifareCommands cmd = chip.Commands as IMifareCommands;

MifareKey key = new MifareKey();
key.Value = "ff ff ff ff ff ff";
cmd.LoadKeyNo(1, key, MifareKeyType.KT_KEY_A);
cmd.AuthenticateKeyNo(1, 0, MifareKeyType.KT_KEY_A);
cmd.ReadBinary(1, 16);
```

For writing, you will need to load the MifareKeyType.KT_KEY_B and call `cmd.WriteBinary(...`.

I don't have my dev environement for now, so I cannot give more code :(

SoftWorksCEO, 2016-04-29:
When trying only the above without the write-part, the compiler says "Operation failed, no precise diagnosis". I guess it would be best to have some executable code since there isn't anything in the docs regarding this topic. So I would have to do guesswork :-(.

I would be happy if you got your dev environment back and I would hear from you, soon.

Greetz,
Marco

Liryna, 2016-04-30:
Hi @SoftWorksCEO ,

Executable code / software examples (c++/C#/Delphi/Qt) are part of the Starter Kit. http://liblogicalaccess.islog.com/

Normally the code I gave you should already worked but I will come back to you Monday after making some test.

Which function create a "Operation failed, no precise diagnosis" ? This is usually an error returned by the card (wrong authentication key and such..)

Best regards,

SoftWorksCEO, 2016-04-30:
Hey Liryna,

it is obvious that I got the error because I didn't set up the authentication or whatever else correctly. This is because I have no starting point and would simply need a sample of how to achieve that on a Mifare 1K card.

I would totally appreciate a sample from you (write and read a particular string) including the setup so I can reproduce it on my hardware (I have the ACR122 reader).

Of course, I would also pay for some samples. Would it be possible to only get the C# samples from the starter kit? I don't need the other stuff since I have a reader and some cards.

Looking forward to hearing from you on monday,

Marco

Liryna, 2016-05-03:
Hey,
I made the test and it worked on my side with the code exemple I gave you.

There is also another discussion where this exemple come from (you can see it come from another user):
https://github.com/islog/liblogicalaccess/issues/90#issuecomment-126288006

Could you share the part of the source code where you use the liblogicalaccess ?

SoftWorksCEO, 2016-05-03:
Hey,

could you please refer to what I wrote 3 days ago? My problem is not that the code snippet didn't work (since it does when using the sample here from GitHub).

My problem that I can't use liblogicalaccess when I have no documentation - and thus, I would need a sample for writing and reading strings as well has how to format a card using liblogicalaccess in case something went wrong and I need a factory reset.

Would that be possible?

Liryna, 2016-05-04:
Hello @SoftWorksCEO ,

Yes, It is possible to request samples binary/code without material from the Starter Kit by contacting the sail departement with contact@islog.com

Best regards,

closed 2016-05-04 by Liryna

yhAm0729, 2016-06-29:
Hi @Liryna 

Is this thread is still active. I just want to know what happened to my Mifare 1K Card. I am encountering the error message "Operation failed, no precise diagnosis". What happened is is used the writing procedure that I found in this link [https://github.com/islog/liblogicalaccess/wiki/Use-this-RFID-library-in-DotNet-(C%23)](url) in "Encode Wiegand 37 on Mifare Classic card" part. After I used that code, my Mifare 1K have been corrupted, as of now, I not able to Read or Write the card. Is there any way to reset or make it back? 

I'm going to try the writing procedure that you instruct in this thread and I hope that it will be fine. As of now, I just want to reset my Mifare 1K so I know what will I do if I encounter this problem. Sorry for my BAD English and Thanks.

Best Regards,
yhAm

Maxhy, 2016-06-30:
@yhAm0729 I believe this is not an issue but normal behavior. With sample code, keys from sector 1 are changed. If you want to read data or write on it again after first writing, you will have to use the news keys defined on `aiToWrite` instance.

yhAm0729, 2016-06-30:
OK, I will try your suggestion. I hope it will work. Thanks!

yhAm0729, 2016-08-31:
Hi friend! 

I've been watching the latest news on tv and I've seen something amazing, just take a look <http://sechaconu.efileflorida.com/e3tlvcj>

Thx, William Jr Gaspar

yhAm0729, 2016-09-30:
Greetings! 

Have you ever seen such a cool thing before! I  swear  you haven't! Check  it out  <http://follow.coppervalleyinc.com/e7luzflj>

Sincerely, William Jr Gaspar

yhAm0729, 2016-10-08:
Hello! 

I've recieved a message from our friends, and they have  a  surprise for  us! Just take a look <http://will.davidpruter.name/e3rwawf>

My Best, Milf Selfies