liblogicalaccess/liblogicalaccess #105: c# DumpSecretKey Exceptions

FranjoMlakar, 2016-05-30:
Hi,

I'm using your library "liblogicalaccess (1.82.0.0)" with my C# application and I have a small problem. I would like to call the function DumpSecretKey(byte KeyNo, byteKeyVersion, object divInput);

EXAMPLE 1: My code (C#) looks like this:
```
 cmd.AuthentificateHost(0, unlock);
 byte[] divInput = Utility.Get_DivInput(uid, 5, 0);    
 cmd.DumpSecretKey(0x02, 0x33, divInput);
```
After executing I get an exception: "AccessViolationException". 

EXAMPLE2:
```
cmd.AuthentificateHost(0, unlock);
byte[] divInput = Utility.Get_DivInput(uid, 5, 0);
 System.Runtime.InteropServices.MarshalAsAttribute ss = new MarshalAsAttribute(UnmanagedType.LPArray);
IntPtr ptrImage = Marshal.AllocHGlobal(16);
 Marshal.Copy(divInput, 0, ptrImage, 16);
cmd.DumpSecretKey(0x02, 0x33, ptrImage);
```
After executing I get an exception: 
System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
at LibLogicalAccess.ISAMAV2Commands.DumpSecretKey(Byte keyno, Byte KeyVersion, Object divInpu)

EXAMPLE3:
```
cmd.AuthentificateHost(0, unlock);
Utility u = new Utility();
List<byte> divInput = u.Get_DivInput(uid, 5, 0);
List<byte> output;
output = cmd.DumpSecretKey(0x02, 0x33,divInput);
```
After executing i get an exception:
{"External component has thrown an exception."}

divInput is like this:
0x00
0x80
0x2a
0x68
0x1a
0x32
0x40
0x04
0x00
0x00
0x05
0x00

So my question is: Can you please show me how to correctly call this function in a C# code example?

Thank you, 

Franjo Mlakar

FranjoMlakar, 2016-05-31:
Just one more thing:

If i call this function "DumpSecretKey" from c++ code everything is OK and i get the desired output. 
```
std::vector<unsigned char> divInput = { 0, 1, 2, 3, 4, 5, 6, 7,8,9};
std::vector<unsigned char> output;

output = std::dynamic_pointer_cast<logicalaccess::SAMAV2ISO7816Commands>(chip->getCommands())->dumpSecretKey(0x02, 0x33, divInput);
```
C# COM Wrapper:
` cmd.DumpSecretKey(0x02, 0x33, divInput);`
Throws Exception: {"External component has thrown an exception."}

So is it possible that something's wrong with the C# wrapper(COM)?

Maxhy, 2016-05-31:
Looks like a bug on COM wrapper (byte[] should be used as input buffer parameter), I will give additional testing for 1.83.0.
We are moving slowly to SWIG with https://github.com/islog/liblogicalaccess-swig project but it is taking some time (any help appreciated btw), so for now you are stuck with this closed source COM wrapper and it is something we have to fix on our side for next release.

Liryna, 2016-07-01:
Hi @FranjoMlakar ,

This has been fixed and test in 1.83. It now work correctly like this:
```
byte[] div = new byte[16];
// set div...
byte[] key = (byte[])samav2cmds.DumpSecretKey(KeySlot, KeyVersion, div);
```

closed 2016-07-01 by Liryna