liblogicalaccess/liblogicalaccess #106: c# AuthentificateHost

FranjoMlakar, 2016-05-31:
Is there a way to specify the HostMode like in c++ function?

c# function:
AuthentificateHost(keyNo, KEY)

c++ function
AuthentificateHost(KEY, keyNo, HostMode)

Maxhy, 2016-05-31:
I'm not sure what you are talking about, C++ function doesn't have HostMode parameter neither. Could you details with github line code link?
Btw there is a typo here, function should be renamed to AuthenticatHost on a further release.

FranjoMlakar, 2016-05-31:
Sorry, my bad. It seems i have some altered source code. In my case there is a function 
`void SAMAV2ISO7816Commands::authentificateHost(std::shared_ptr<DESFireKey> key, unsigned char keyno, HostMode ihostmode)`

closed 2016-05-31 by FranjoMlakar

Maxhy, 2016-05-31:
Ok good to know.
I believe you are doing right, but don't forget to respect [LGPL license](https://github.com/islog/liblogicalaccess/blob/master/LICENSE.LGPL.txt) about source code modification, program publication and referenced libraries on your program copyright notice. Thanks.

FranjoMlakar, 2016-05-31:
Don't worry. I'm just trying to learn how all this things are working. Will respect the LGPL license. Thanks for your answer. 
best regards,