liblogicalaccess/liblogicalaccess #107: System.Runtime.InteropServices.COMException When continuously reading cards

ishara0925, 2016-07-23:
Hi,
I have run the sample code with ACR122U reader for a MiFare1K card. It work without error as long as I read one card and terminate the program. But when I try to read cards inside a while loop so that program doesn't exit when the card removed. It will wait for a another card to be inserted. But when it comes to the second card it gives me the following exception. how can I solve this?

[image]
thank you

Liryna, 2016-07-23:
Hi @ishara0925 ,

Can you copie past all your code with the while ?

ishara0925, 2016-07-23:
Thank you for the fast reply .here you go.
`       static void Main(string[] args)
        {
            // Explicitly use the PC/SC Reader Provider.
            IReaderProvider readerProvider = new PCSCReaderProvider();

            // Create the default reader unit. On PC/SC, we will listen on all readers.
            IReaderUnit readerUnit = readerProvider.CreateReaderUnit();
            byte[] data = new byte[] { 0x00, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 
                    0x00, 0x00, 0x00, 0x62, 0x03, 0x01, 0x0C, 0x06, 0x01 };

            while (true)
            {
                chip card = connectCard(readerUnit);
                
                string line;

                List<string> rrData = new List<string>();
              //  rrData = readCardBlock(card, 3, 0);
                List<List<string>> x = readCard(card);
                Console.WriteLine("Read[r] or Write[w] =  "); 
                /*while (true)
                {
                    line = Console.ReadLine(); // Get
                    if (string.Equals(line, "w") || string.Equals(line, "r")) break;
                }
                if (string.Equals(line, "w")) writeCard(card, 3, 0, data);
                else if (string.Equals(line, "r")) rrData = readCardBlock(card, 3, 0);*/
                disconnectCard(readerUnit);
            }

        }
            
        
        static void disconnectCard(IReaderUnit tempReaderUnit)
        {
            Console.WriteLine("please Remove the card");
            if (tempReaderUnit.WaitRemoval(0))
            {
                Console.WriteLine("Card removed");

                tempReaderUnit.Disconnect();
                tempReaderUnit.DisconnectFromReader();
            }
        }
  static chip connectCard(IReaderUnit tempReaderUnit)
        {
            chip chipT = new chip();
            if (tempReaderUnit.ConnectToReader())
                {                              
                        try
                        {
                        Console.WriteLine("Insert Card...");
                        if (tempReaderUnit.WaitInsertion(0))
                        {
                            if (tempReaderUnit.Connect())
                            {
                                Console.WriteLine("Card inserted on reader '{0}.'", tempReaderUnit.ConnectedName);
                                chipT = tempReaderUnit.GetSingleChip();
                                Console.WriteLine("Card type: {0}", chipT.Type);
                                Console.WriteLine("Card unique manufacturer number: {0}", tempReaderUnit.GetNumber(chipT));

                                if (chipT is Mifare1KChip)
                                {
                                    Console.WriteLine("Mifare1K is Detected");
                                }
                                else
                                {
                                    Console.WriteLine("Not Mifare1K card.Please Insert the correct card");
                                }
                            }
                        }
                        }catch(Exception e){
                            Console.WriteLine(e.Message);
                        }
            }
            return chipT;                  
        }
static List<List<string>> readCard(chip chip)
        {
            IMifareCommands cmd = chip.Commands as IMifareCommands;

            MifareKey key = new MifareKey();
            key.Value = "ff ff ff ff ff ff";
            cmd.LoadKeyNo(1, key, MifareKeyType.KT_KEY_A);
            cmd.AuthenticateKeyNo(1, 0, MifareKeyType.KT_KEY_A);
            cmd.ReadBinary(1, 16);

            SectorAccessBits sab = new SectorAccessBits();
            sab.d_data_block0_access_bits.c1 =1;
            sab.d_data_block0_access_bits.c2 = 0;
            sab.d_data_block0_access_bits.c3 = 0;

            for (int i = 0; i < 16; i++)
            {
                (chip.Profile as IMifareProfile).SetKey(i, MifareKeyType.KT_KEY_A, "ff ff ff ff ff ff");
            }

            var list = new List<List<string>>();

            for (int i = 0; i < 16; i++)
            {
                try
                {
                    object data = cmd.ReadSector(i, sab, false);
                    IEnumerable enumerable = data as IEnumerable;
                    if (enumerable != null)
                    {
                        var listChar = new List<string>();

                        foreach (object element in enumerable)
                        {
                            int intValue = Convert.ToInt32(element.ToString(), 10);
                            char c = Convert.ToChar(intValue);
                            string hex = intValue.ToString("X");
                            Console.Write(hex + " ");
                            listChar.Add(hex);
                        }
                        Console.WriteLine("");
                        list.Add(listChar);
                        var completeString = string.Concat(list);
                    }
                }
                catch (Exception ex)
                {
                    if (ex.Message == "No key found.")
                    {
                        Debug.WriteLine("No key found for sector: " + i);
                    }
                    else
                    {
                        throw;
                    }
                }
            }
            return list;
        }
    }`

Liryna, 2016-07-23:
@ishara0925 , can you move the ConnectToReader and DisconnectFromReader outside the while ?
This should only be made at the start and at the end of your job.

ishara0925, 2016-07-23:
I did. But the error still comes. Now it is looked like this.

```
static void Main(string[] args)
        {
            // Explicitly use the PC/SC Reader Provider.
            IReaderProvider readerProvider = new PCSCReaderProvider();

            // Create the default reader unit. On PC/SC, we will listen on all readers.
            IReaderUnit readerUnit = readerProvider.CreateReaderUnit();
            byte[] data = new byte[] { 0x00, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 
                    0x00, 0x00, 0x00, 0x62, 0x03, 0x01, 0x0C, 0x06, 0x01 };
            if (readerUnit.ConnectToReader())
            {
                while (true)
                {
                    chip card = connectCard(readerUnit);

                    string line;

                    List<string> rrData = new List<string>();
                    //  rrData = readCardBlock(card, 3, 0);
                    List<List<string>> x = readCard(card);
                    Console.WriteLine("Read[r] or Write[w] =  ");
                    /*while (true)
                    {
                        line = Console.ReadLine(); // Get
                        if (string.Equals(line, "w") || string.Equals(line, "r")) break;
                    }
                    if (string.Equals(line, "w")) writeCard(card, 3, 0, data);
                    else if (string.Equals(line, "r")) rrData = readCardBlock(card, 3, 0);*/
                    disconnectCard(readerUnit);
                }
            }
            readerUnit.DisconnectFromReader();

        }
            
        
        static void disconnectCard(IReaderUnit tempReaderUnit)
        {
            Console.WriteLine("please Remove the card");
            if (tempReaderUnit.WaitRemoval(0))
            {
                Console.WriteLine("Card removed");

                tempReaderUnit.Disconnect();
                
            }
        }
        static chip connectCard(IReaderUnit tempReaderUnit)
        {
            chip chipT = new chip();                             
                        try
                        {
                        Console.WriteLine("Insert Card...");
                        if (tempReaderUnit.WaitInsertion(0))
                        {
                            if (tempReaderUnit.Connect())
                            {
                                Console.WriteLine("Card inserted on reader '{0}.'", tempReaderUnit.ConnectedName);
                                chipT = tempReaderUnit.GetSingleChip();
                                Console.WriteLine("Card type: {0}", chipT.Type);
                                Console.WriteLine("Card unique manufacturer number: {0}", tempReaderUnit.GetNumber(chipT));

                                if (chipT is Mifare1KChip)
                                {
                                    Console.WriteLine("Mifare1K is Detected");
                                }
                                else
                                {
                                    Console.WriteLine("Not Mifare1K card.Please Insert the correct card");
                                }
                            }
                        }
                        }catch(Exception e){
                            Console.WriteLine(e.Message);
                        }
            return chipT;                  
        }
```

ishara0925, 2016-07-26:
Hi,
I tried the same thing with a timer interrupt in C# Form Application. It also gives me the same error can't read the card twice. Could this be a library issue?
Thank you

Liryna, 2016-07-26:
it is probably the behaviour of the reader with load key.
https://github.com/islog/liblogicalaccess/blob/1e6af52b1c45ab55e3b21a4cd92ac2f6227cc93f/plugins/pluginsreaderproviders/pcsc/commands/mifarepcsccommands.cpp#L50

Are you able to reproduce the code in c++ and see which part is failing here ?

xaqq, 2016-07-26:
Hello,
I have a partial reproduction using C++. Actually, I don't even go as far as you did, because I get "Operation Failed" as soon as I try to load a key (even the first time).

This is indeed related to what @Liryna linked. The code works correctly when we force to `volatile` memory. My local fixe consists of editing acsacrresultchecker and marking the `{0x63, 0x00}` error code as `CardException::UNKOWN_ERROR`. This way, the current check in mifarepcsccommands works correctly.
We may need to check that there are no additional side effect wrt changing error code.

Maxhy, 2016-07-26:
We may probably load the key in volatile memory by default btw instead of non-volatile, in most use case you don't want to store it on the reader forever...

ishara0925, 2016-07-27:
hi,
Do I have to rebuild the source project to get lib files?

Maxhy, 2016-07-28:
Yes you will have to rebuild to test it.
Otherwise if you don't want to rebuild yourself then you will have to wait for the next release in few weeks.

Maxhy, 2016-08-22:
@ishara0925 a new release was just published which includes this fix.

closed 2016-09-14 by Maxhy