liblogicalaccess/liblogicalaccess #108: GetFileSettings crashes for cyclic record file on DESFire EV1

PeVoo, 2016-09-13:
Hi,

The method IDESFireEV1Commands.GetFileSettings crashes if used on a cyclic record file. I'm using C# and COM object 1.83, but the error is unmanaged and cannot be even caught by VS, so it must be in the library (or COM). Please help, it's urgent. 

Example application to reproduce the error:

```
using System;
using System.Collections;
using System.Collections.Generic;
using System.Linq;
using System.Runtime.InteropServices;
using System.Text;
using System.Threading.Tasks;

namespace BugReportApp
{
  class Program
  {
    static void Main(string[] args)
    {
      try
      {
        IReaderUnit reader = null;
        chip chip;
        IDESFireEV1Commands cmd;

        var readerProvider = new PCSCReaderProvider();
        IEnumerable readerList = readerProvider.GetReaderList();

        IEnumerable found = (IEnumerable)readerProvider.WaitForReaders(new string[] { "OMNIKEY CardMan 5x21-CL 0" }, 1, true);
        foreach (IReaderUnit item in found.OfType<IReaderUnit>())
          reader = item;  // this will be max 1 iteration

        if (reader == null)
          throw new Exception("Can't find the reader!");

        if (!reader.ConnectToReader())
          throw new Exception("Can't connect to the reader!");

        reader.WaitInsertion(1000);
        if (!reader.Connect())
          throw new Exception("Can't connect to the card!");

        chip = reader.GetSingleChip();
        if (!(chip is IDESFireChip))
          throw new Exception("This is not a DESFire EV1 card!");

        cmd = chip.Commands as IDESFireEV1Commands;
        if (cmd == null)
          throw new Exception("This is not a DESFire EV1 card!");

        DESFireKey masterKey = new DESFireKey();
        masterKey.Value = "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00";

        cmd.SelectApplication(0);
        cmd.Authenticate(0, masterKey);

        cmd.Erase();
        cmd.CreateApplicationEV1(10, DESFireKeySettings.KS_DEFAULT, 4, false, DESFireKeyType.DF_KEY_AES, 0, null);
        cmd.SelectApplication(10);

        DESFireKey key0 = new DESFireKey();
        key0.Value = "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00";
        key0.KeyType = DESFireKeyType.DF_KEY_AES;
        cmd.Authenticate(0, key0);

        key0.Value = "AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA";
        cmd.ChangeKey(0, key0);
        cmd.Authenticate(0, key0);

        DESFireKey key1 = new DESFireKey();
        key1.Value = "11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11";
        key1.KeyType = DESFireKeyType.DF_KEY_AES;
        cmd.ChangeKey(1, key1);

        DESFireKey key2 = new DESFireKey();
        key2.Value = "22 22 22 22 22 22 22 22 22 22 22 22 22 22 22 22";
        key2.KeyType = DESFireKeyType.DF_KEY_AES;
        cmd.ChangeKey(2, key2);

        Console.WriteLine("Created application 0x0A with 2 additional keys.");

        DESFireAccessRights access;
        access.readAccess = TaskAccessRights.AR_KEY1;
        access.writeAccess = TaskAccessRights.AR_KEY2;
        access.readAndWriteAccess = TaskAccessRights.AR_KEY2;
        access.changeAccess = TaskAccessRights.AR_KEY2;

        cmd.CreateStdDataFile(1, EncryptionMode.CM_ENCRYPT, access, 10);
        Console.WriteLine("Created standard data file: 1");
        cmd.CreateCyclicRecordFile(2, EncryptionMode.CM_ENCRYPT, access, 10, 2);
        Console.WriteLine("Created cyclic record file: 2");

        byte[] testData = new byte[] { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10 };
        cmd.Authenticate(2, key2);

        cmd.WriteData(1, 0, 10, EncryptionMode.CM_ENCRYPT, testData);
        Console.WriteLine("Saved data to file 1.");
        cmd.WriteRecord(2, 0, 10, EncryptionMode.CM_ENCRYPT, testData);
        Console.WriteLine("Saved data to file 2.");

        byte[] files = cmd.GetFileIDs();
        Console.WriteLine("\r\nFiles reported on card: " + string.Join(", ", files.Select(f => ((int)f).ToString())));

        foreach (byte file in files)
        {
          Console.Write("Type of file " + file + ": ");
          var settings = cmd.GetFileSettings(file);
          Console.WriteLine(settings.FileType);
        }
      }
      catch (COMException ex)
      {
        Console.WriteLine("Reader exception: " + ex.Message);
      }
      catch (Exception ex)
      {
        Console.WriteLine("Unexpected exception: " + ex);
      }
      Console.WriteLine("Finished.");
      Console.ReadLine();
    }
  }
}
```

Application's output: [screen here](https://postimg.org/image/t5km3srgv/)

> Created application 0x0A with 2 additional keys.
> Created standard data file: 1
> Created cyclic record file: 2
> Saved data to file 1.
> Saved data to file 2.
> 
> Files reported on card: 1, 2
> Type of file 1: 0
> Type of file 2: 

As you can see, the first file works fine, and the cyclic one crashes ("vshost32.exe has stopped working"). 
It happens every time. Thank you for help.

PeVoo, 2016-09-13:
**This is from the Event Viewer:**

> Faulting application name: BugReportApp.vshost.exe, version: 14.0.23107.0, time stamp: 0x559b788a
> Faulting module name: liblogicalaccess.com.dll, version: 1.83.704.705, time stamp: 0x577a5bf0
> Exception code: 0xc0000409
> Fault offset: 0x000e2e3e
> Faulting process ID: 0x2678
> Faulting application start time: 0x01d20dca3104eec2
> Faulting application path: C:\Projects\BugReportApp\bin\Debug\BugReportApp.vshost.exe
> Faulting module path: C:\Program Files (x86)\Common Files\ISLOG\LibLogicalAccess\1.83.1.0704\liblogicalaccess.com.dll
> Report ID: 6544613b-dbf1-4a9b-9110-20e76366f622
> Faulting package full name: 
> Faulting package-relative application ID:

Maxhy, 2016-09-13:
This looks like a bug indeed, probably on the COM wrapper. I'm going to fix it when I get some times for it, thanks for the report! Fix will be provided on 1.85 release planned around October.

silverfrost, 2016-09-30:
Have you tried redoing an Authenticate before calling GetFileSettings?

PeVoo, 2016-10-03:
Yes, my file checking loop selects the application and authenticates with the app's master key (0) before each file's settings are read. Other files read correctly, only the cyclic crashes. And it crashes in non-catchable way, so I can't even try to check the app's files without the risk of crashing my app. And that's really bad. I had to write a really ugly workaround.

Maxhy, 2016-10-06:
Tested on 1.85.0 with your sample code successfully!

closed 2016-10-06 by Maxhy