liblogicalaccess/liblogicalaccess #109: ACR122 supported

guillerot, 2016-09-13:
Hi,

On the supported hardware page, it is not clear which ACR122 is supported (ACR122U/ACR122S)
Are they both supported ?
Does this support require the use of liblogicalacccess-libnfc ?

Regards,
Adrien G.

(PS: merci pour cette superbe lib)

Maxhy, 2016-09-13:
ACR122U is supported by default with liblogicalaccess over PC/SC. If you want you can also use it through libnfc using liblogicalaccess-libnfc but there is generally no real requirement for that here.
ACR122S is not supported. I believe the command set is the same so if this is what you're looking for, a new ACR122S reader plugin can be easily created on liblogicalaccess that use same command classes on a new reader provider, reader unit and data transport (inherited from serial port).

(PS: merci pour l'encouragement. N'hésites pas à nous faire part de ton projet, cela fait toujours plaisir de savoir où l'on a été util :wink:).

guillerot, 2016-09-13:
I will see if I can work on a new plugin for the ACR122S. Not sure if it will be easy for me, my first language is C# (and that's why liblogicalaccess interest me, because there is a wrapper already available :smile: )

guillerot, 2016-09-14:
So based on the ACR122S communication protocol given on this page:
http://www.acs.com.hk/en/driver/121/acr122s-serial-nfc-reader/
+ existing classes using SerialPortDataTransport (ie: stidstrreaderdatatransport)
I should be able to add easily a new plugin ?

Maxhy, 2016-09-14:
Yes indeed that's the card, but you will also need to create a reader provider and reader unit for your reader (probably generic to all ACS Serial Port readers). Once you're ready, don't hesitate to suggest your PR and I will gladly review it.

I don't have this reader so I cannot help you, but if your project is for personal use or open source project then simply offer us a ACR122S reader and I will implement it into liblogicalaccess when I get some free time (not much right now but probably more in few weeks). If your project is for business you may be interested to know that we can also integrate some specific reader or chip under development contract (at a preferential price as it is for an open source project, to promote such decision positive for the global community).

Maxhy, 2016-09-16:
Closing this as it's not a real issue, but feel free to continue the discussion if you want to.

closed 2016-09-16 by Maxhy

guillerot, 2016-09-16:
Thanks for your time. I will continue the discussion if I start to work on the plugin.