liblogicalaccess/liblogicalaccess #110: Windows Service

m8kwr, 2016-09-16:
I am getting the following error when using code when trying to use the example .net code in a windows service.

> System.OutOfMemoryException: Creating an instance of the COM component with CLSID {707D1804-5D33-4BFA-984A-EFCB14AECB1A} from the IClassFactory failed due to the following error: 8007000e Not enough storage is available to complete this operation. (Exception from HRESULT: 0x8007000E (E_OUTOFMEMORY)).

It errors on the following code:

`// Explicitly use the PC/SC Reader Provider.
IReaderProvider readerProvider = new PCSCReaderProvider();`

Liryna, 2016-09-16:
@m8kwr Hi, 

This mean the installation of liblogicalaccess didn't work. 
Please try to reinstall

m8kwr, 2016-09-16:
Many thanks.  I think there is an issue with my machine, as all worked as expected on another.

closed 2016-09-16 by Maxhy