liblogicalaccess/liblogicalaccess #111: Mifare authenticate answer incorrect for OK 5553

guillerot, 2016-09-20:
Hi,

From what I've read here : http://www.nfc-webshop.com/media/wysiwyg/download/rfid/rfid_reader/hf/iso15693/protocol/TAGscan003_UserManual_HF_Multi_ISO_RFID_1.pdf

- 'L' Login success
- 'F' Error: General failure
- 'N' Error: No tag
- 'O' Error: Operation mode failure
- 'R' Error: Out of range
- 'X' Error: Authentication failed 

I might be wrong but test on answer seems incorrect for MifareOK5553Commands::authenticate
Should be :
```cpp
if (answer[0] != 'L')
{
    THROW_EXCEPTION_WITH_LOG(LibLogicalAccessException, "Authentication error.");
}
```

Regards.

Maxhy, 2016-09-22:
Indeed this looks incorrect. Even if it is a small change, could you suggest a PR so credits will go to you on git history :smile:?
Otherwise I would not recommend to use this reader for new projects as it is deprecated by the manufacturer.

guillerot, 2016-09-22:
I will.
We do not use this reader. I was looking for inspiration before working on a ACS ACR122S Plugin (and maybe an ACR1281S).

commit 7d748d1d (2016-09-30): Fix for Mifare Authenticate against OK 5553.

This fixes #111.
closed 2016-09-30 by xaqq