liblogicalaccess/liblogicalaccess #112: OutOfMemoryException when trying to initialize PCSCReaderProvider

billylukito, 2016-09-22:
Hi,

I'm trying to initialize a PCSCReaderProvider object but OutOfMemoryException occurs.

This is the code that produces the error:

` IReaderProvider readerProvider = new PCSCReaderProvider();`

and this is the error message:

> An unhandled exception of type 'System.OutOfMemoryException' occurred in mscorlib.dll

> Additional information: Creating an instance of the COM component with CLSID {707D1804-5D33-4BFA-984A-EFCB14AECB1A} from the IClassFactory failed due to the following error: 8007000e Not enough storage is available to complete this operation. (Exception from HRESULT: 0x8007000E (E_OUTOFMEMORY)).

Could you help me what to do with this error?

Thanks,
Billy

Liryna, 2016-09-22:
Hi @billylukito ,

Does the smart card service is started on your windows ?
Also, does the library is correctly installed ?

billylukito, 2016-09-23:
It looks like the service is the problem.
The OutOfMemoryException solved when the service runs.

Thank you very much @Liryna .

closed 2016-09-23 by Liryna