liblogicalaccess/liblogicalaccess #113: NXP RD720 card reader doesn't detect any cards

PeVoo, 2016-09-26:
I've got a new reader NXP RD720, with its original drivers from NXP's site. It is detected by the library and reported as "NXP Pegoda N CL 0 0". In the Device Manager it is reported as "NXP Pegoda USB" under "Smart card readers". 

The WaitInsertion method always returns false and no card is ever reported. The reader is OK, because I can both detect and read cards with NXP's RFIDDiscoverLite software. Is there something I'm missing or is it a bug?

Liryna, 2016-09-26:
Hi @PeVoo ,

Does the software readcard http://www.islog.com/logiciels/readcard/ succeed to show the card ?

PeVoo, 2016-09-26:
@Liryna Nope, not on this reader. The Omnikey reader works fine (both in readcard and my app).

Maxhy, 2016-09-30:
Mhh this is weird as it should work the same for all PC/SC readers (it is at least the case for the ~30 PC/SC readers I tested until today). I don't know about NXP RD720 reader but previous models NXP RD701 wasn't PC/SC compliant. This one looks indeed to support PC/SC according to the specification. Maybe it is a reader behavior conflicting with LLA implementation. Feel free to manually debug it using C sample, or I will gladly debug it for you if I have a reader.

PeVoo, 2016-10-03:
I'm sorry but I'm using the C# COM version, so I cannot debug your code. Also I had to give the reader back to its owner, so I don't have it anymore. The only option now is that you get one for your tests.

Liryna, 2016-10-03:
The reader seems to be deprecated. It is gonna be hard and expensive to find one.

Maxhy, 2016-10-04:
I agree. I will close this issue with`wontfix` as chances that it will be fixed is low now. If someone want a fix for this reader, please offer such reader to Liryna or myself.

closed 2016-10-04 by Maxhy