liblogicalaccess/liblogicalaccess #115: What do I need to include to distribute LobLogicalAccess?

silverfrost, 2016-09-29:
I have some code that works very nicely in .NET with liblogicalaccess. I want to ship the code -- what do I need to send?  I could the the full liblogicalaccess but that is many times the size of my app :) Presumably I need to ship the core files and the drivers DLLs for hardware.  

Which are the core files? Do I have to regsvr some of them?

Liryna, 2016-09-29:
If you use .NET you will need to use the full installer because the .NET link with all libraries.

Maxhy, 2016-09-30:
Indeed, and it will be the same for the coming [LibLogicalAccess SWIG wrapper](https://github.com/islog/liblogicalaccess-swig).
Ideally the SWIG wrapper will be flexible and works also with plugins but that's not the case at current state (open for suggestion)..

closed 2016-09-30 by Maxhy

silverfrost, 2016-09-30:
Okay. I use liblogicalaccess for DESFire, and I suspect there will be fairly few (numbers of ) users. So I might just detect if it is there and tell them to install it separately.

So if the DLLs that I am not using are not there the .NET stuff will not work?

Afterster, 2016-09-30:
Unfortunately that's right when using .NET, all dll need to be deployed because the COM layer link with it.
Btw, you should use LibLogicalAccess.exe installer that is provided at each release which also check for VC++ runtime dependencies, you can install it in silent mode with `/quiet`.

Feel free to share a link to your project/product as it is always interesting to know where LLA is helping ;)

Maxhy, 2016-09-30:
Unfortunately that's right when using .NET, all dll need to be deployed because the COM layer link with it.
Btw, you should use LibLogicalAccess.exe installer that is provided at each release which also check for VC++ runtime dependencies, you can install it in silent mode with /quiet.

Feel free to share a link to your project/product as it is always interesting to know where LLA is helping ;)

silverfrost, 2016-09-30:
It is in this product (or will be): https://www.easybadge.com/

Maxhy, 2016-09-30:
This looks nice! 👍 

Don't forget the legal open source credits mention ;). Just kidding (even if true :smile:) but any reference to ISLOG or LibLogicalAccess somewhere would be appreciated if it helps your business.

I will also promote ours here as I know LLA is mainly used on commercial and closed source software (general talking, please don't take it personally): regular donations are appreciated as it may be a way to guarantee the development team and project good health. Which is far from perfect today as LLA never got any donation at all since we decided to put it open source in 2013 ; the project is only alive thanks to ISLOG company. When ISLOG competitors are using LLA on their ISLOG similar products, this is fine from the license and LLA maintainers (tech guys :smile:) point of view, but if they don't financially support the project (donation, contribution, partnership, maintenance plans...) when they decrease the business of its main supporter, the dev team will be in danger ; which means LLA project itself may be harmed. It's not the case today, just warning for the long run.

In short, for those doing business around LLA: be fair.