liblogicalaccess/liblogicalaccess #117: SAM Detected is not the SAM waited

cxminer, 2016-10-10:
With:
```
var readerUnitConfig = (PCSCReaderUnitConfiguration)readerUnit.Configuration;
readerUnitConfig.SAMReaderName = samReaderName; // samReaderName=Identive CLOUD 4710 F SAM Reader 0
readerUnitConfig.SAMType = "SAM_AV2";
readerUnitConfig.setSAMUnLockKey(1, unlockkey);
```
I get an error saying "SAM Detected is not the SAM waited" when calling `readerUnit.ConnectToReader()`.

I don't know what I am missing here. The SAM reader clearly is the SAM waited as I got the SAM reader name from calling `new PCSCReaderProvider().GetReaderList();`.

_I know this is more of a forum kind of a post, but since there is no such thing I decided to post here._

Maxhy, 2016-10-10:
I believe you're using  a NXP SAM AV2 chip here? Be aware that these chips are provided in SAM AV1 mode by default and you have to switch to AV2 mode if you want to use AV2 features.
If you don't know about that, please take a look at NXP documentation or use a SAM production software (eg. [SAM Manager](http://www.islog.com/en/software/sam-manager/)) to pre-initialize your SAMs.

Indeed there is no such things like LLA forum or group, so no problem. There was but we closed it as we want everything to be public and on the same place (except for companies under support plan which have a private communication channel with ISLOG development team). Hope it helps ; *thanks* are welcomed :wink:.

closed 2016-10-10 by Maxhy

cxminer, 2016-10-11:
Thank you for your response and help.

Yes I am using SAM AV2 chip but we got it from a provider which I believe already initialized SAM to AV2 mode. Also when I try to run the following snippet:

```
var provider = new PCSCReaderProvider();
var reader = (PCSCReaderUnit)provider.CreateReaderUnit();
reader.SetName(samReaderName);

if (reader.ConnectToReader())
{
	if (reader.WaitInsertion(10000))
	{
		if (reader.Connect())
		{
			var chip = reader.GetSingleChip();

			if (chip != null)
			{
				var commands = chip.Commands as SAMAV1Commands;
				commands.LockUnlock(unlockkey, SAMLockUnlock.SwitchAV2Mode, 1, 1, 1);
			}
		}
	}
}
```

I get an error "Object reference not set to an instance of an object." because `chip.commands` cannot be cast to `SAMAV1Commands`, however if I change this to `SAMAV2Commands` it works (the call to `LockUnlock` fails, but I get the commands), isn't this also in indicator that SAM is initialized in AV2 mode?

Liryna, 2016-10-11:
Hi @cxminer ,

What is the error returned by LockUnlock ?

cxminer, 2016-10-11:
I get `conditions of use not satisfied, invalid key type, invalid CID, key limit reached`. However I am sure that my key is correct. Here is the code snippet for my key:

```
var unlockkey = new DESFireKey();
unlockkey.KeyType = DESFireKeyType.DF_KEY_AES;
unlockkey.Value = "00000000000000000000000000000000";
unlockkey.KeyVersion = 1;
```

The value of my key is different of course, but same in length. _(can't really post it here)_

Liryna, 2016-10-11:
@cxminer you need to check the configuration of your keyentry and find out if you have a right to perform the lockunlock request on it and also the current state of your SAM.

cxminer, 2016-10-11:
Let me see if I got this right @Liryna. by _"check the configuration of your keyentry"_ you mean the `KeyEntryInformation` property? I tried that and I figured out that I have 3 keys in there (I assume those are the 3 diversified keys) at position 0x00, 0x01 and 0x02 however I am not clear on what the fields in `s_KeyEntryAV2Information` mean so I will need to investigate this further.

About _"if you have a right to perform the lockunlock request on it"_, how can I check that? Also what do you mean by _"current state of your SAM"_? 

Any further help would be greatly appreciated.

Liryna, 2016-10-11:
@cxminer This informations is in the NXP documentation that is under NDA. I will not be able to provide it unfortunately 😢 .
The SAM is a difficult hardware to handle and liblogicalaccess offer an easy way to use his functionality with a minimum of knowledge on it but still needed.

I would like to help you more on this but do not see how to do it without breaking the NDA that we signed with NXP.

cxminer, 2016-10-11:
Ah yes I see, well, thank you for all your help. Our company also signed the NDA with NXP and are waiting for approval, I just thought I can figure it out without the proprietary documentation but I guess not. 😆 

Much appreciate all your help @Liryna and @Maxhy.

Maxhy, 2016-10-11:
You're welcome! I hope you will found liblogicalaccess useful, please feel free to share your company name for the record.

We hesitate to provide some "LLA Users Directory" where users could register themselves and have access to more information and support according to several criteria (confirmed LLA advertising, confirmed NXP partner, ISLOG contracted support plan, ...). This idea originally come following latest https://github.com/islog/liblogicalaccess/issues/115#issuecomment-250753406 discussion.

What do you think?