liblogicalaccess/liblogicalaccess #118: WaitInsertion seems to connect to the wrong reader

silverfrost, 2016-10-26:
In my .NET app I use (with version 1.85):

            _ReaderUnit = _ReaderProvider.CreateReaderUnit(  );
            if (_ReaderUnit.ConnectToReader( ))
            {
                if (_ReaderUnit.WaitInsertion( 5000 ))
                {
                    if (_ReaderUnit.Connect( ))

However, when the card arrives at the encoder in the printer it chooses the wrong encoder and produces:

"A first chance exception of type 'System.Runtime.InteropServices.COMException' occurred in printcard.exe

Additional information: The ATR length must be at least 2 bytes long (TS + T0)."

The reader it selects is "OMNIKEY CardMan 5x21 0" when I expected it to be "OMNIKEY CardMan 5x21-CL 0". This does work with a standalone card reader.  Is there anyway I can force the reader to be the CL one??

Maxhy, 2016-11-04:
Indeed this is because your application probably communicates with the wrong reader. You should browse all available readers (`_ReaderProvider.GetReaderList`) and explicitly use the expected reader unit instance.

silverfrost, 2016-11-06:
I was fooled by the fact that the returned objects from GetReaderList just look like COM Objects -- with no obvious way of using them. When I did

object readers = (_ReaderProvider.GetReaderList) as object[]);
_ReaderUnit = readers[0] as IReaderUnit;

I could then see the reader unit. I'll try that on my odd setup.

closed 2016-11-07 by Maxhy