liblogicalaccess/liblogicalaccess #119: Gentoo Build

maxim-smirnov, 2016-11-07:
After building and installing library, nor return available plugins. 
Return of sample program: 
`Available reader plug-ins (0):
Available card plug-ins (0):
Please type the reader plug-in to use:`

After printing anything and pressing enter it crashes with segmentation fault

Liryna, 2016-11-07:
hello @maxim-smirnov ,

It seems like the liblogicalaccess is not able to find where are your plugins.
Can you copy https://github.com/islog/liblogicalaccess/blob/master/liblogicalaccess.config near you application and add here https://github.com/islog/liblogicalaccess/blob/master/liblogicalaccess.config#L27 the folder of plugin ?

maxim-smirnov, 2016-11-07:
Hi^ thank you for fast reply. I've already tried many combinations of <PluginFolder> like /usr/local/lib, /url/lib etc. Now I've launched program with enabled log to cerr. Here it is: http://pastebin.com/Zpj7iYgi

It says there is no plugins in /usr/local/lib, here is content of /usr/local/lib folder:
cmake                              liblegicprimecards.so.1.85.0
liba3mlgm5600readers.so            liblla-tests.so
liba3mlgm5600readers.so.1.85.0     liblogicalaccess-cryptolib.a
libadmittoreaders.so               liblogicalaccess.so
libadmittoreaders.so.1.85.0        liblogicalaccess.so.1.85.0
libaxesstmc13readers.so            libmifarecards.so
libaxesstmc13readers.so.1.85.0     libmifarecards.so.1.85.0
libaxesstmclegicreaders.so         libmifarepluscards.so
libaxesstmclegicreaders.so.1.85.0  libmifarepluscards.so.1.85.0
libcps3cards.so                    libmifareultralightcards.so
libcps3cards.so.1.85.0             libmifareultralightcards.so.1.85.0
libdeisterreaders.so               libok5553readers.so
libdeisterreaders.so.1.85.0        libok5553readers.so.1.85.0
libdesfirecards.so                 libosdpreaders.so
libdesfirecards.so.1.85.0          libosdpreaders.so.1.85.0
libelatecreaders.so                libpcscreaders.so
libelatecreaders.so.1.85.0         libpcscreaders.so.1.85.0
libem4102cards.so                  libpromagreaders.so
libem4102cards.so.1.85.0           libpromagreaders.so.1.85.0
libem4135cards.so                  libproxcards.so
libem4135cards.so.1.85.0           libproxcards.so.1.85.0
libepasscards.so                   libproxlitecards.so
libepasscards.so.1.85.0            libproxlitecards.so.1.85.0
libfelicacards.so                  librplethreaders.so
libfelicacards.so.1.85.0           librplethreaders.so.1.85.0
libgenerictagcards.so              libsamav2cards.so
libgenerictagcards.so.1.85.0       libsamav2cards.so.1.85.0
libgigatmsreaders.so               libscielreaders.so
libgigatmsreaders.so.1.85.0        libscielreaders.so.1.85.0
libgunneboreaders.so               libseoscards.so
libgunneboreaders.so.1.85.0        libseoscards.so.1.85.0
libicode1cards.so                  libsmartframecards.so
libicode1cards.so.1.85.0           libsmartframecards.so.1.85.0
libicode2cards.so                  libsmartidreaders.so
libicode2cards.so.1.85.0           libsmartidreaders.so.1.85.0
libidondemandreaders.so            libstidprgreaders.so
libidondemandreaders.so.1.85.0     libstidprgreaders.so.1.85.0
libindalacards.so                  libstidstrreaders.so
libindalacards.so.1.85.0           libstidstrreaders.so.1.85.0
libinfineonmydcards.so             libstmlri512cards.so
libinfineonmydcards.so.1.85.0      libstmlri512cards.so.1.85.0
libiso15693cards.so                libtagitcards.so
libiso15693cards.so.1.85.0         libtagitcards.so.1.85.0
libiso7816cards.so                 libtopazcards.so
libiso7816cards.so.1.85.0          libtopazcards.so.1.85.0
libiso7816readers.so               libtwiccards.so
libiso7816readers.so.1.85.0        libtwiccards.so.1.85.0
liblegicprimecards.so

Liryna, 2016-11-07:
😮 He should normally find them 

Can you add more logs here https://github.com/islog/liblogicalaccess/blob/master/src/dynlibrary/librarymanager.cpp#L252
He seems like to not pass this if

maxim-smirnov, 2016-11-07:
Rebuilding lib? 😄 Sweet)

Liryna, 2016-11-07:
😢 yes, but this is very strange that boost return that the path does not exist/not a folder.
Does your user have access to the folder ?

maxim-smirnov, 2016-11-07:
So, LOL. It says that directory is'n exist:

2016-Nov-07 20:27:25.503767 - PLUGIN: 	{252}	{scanPlugins}	{/home/user/liblogicalaccess/src/dynlibrary/librarymanager.cpp}:
For "/usr/local/lib"
Is existing: 0
Is it a directory: 0

Yes, user have access. May be it's problem with symlinks?

maxim-smirnov, 2016-11-07:
Although, I've tried to run program with root. No effect.
Just tried with lib64 (without symlinks). No effect.

Boost version 1.62.0

Liryna, 2016-11-07:
We are using debian/ubuntu versions here 1.54/1.55 but this is impressive if boost broke it 😮

maxim-smirnov, 2016-11-07:
It's printing pluginDir variable with quotes at the beginning and at the end of string. Is it normal?

Liryna, 2016-11-07:
There should be not quotes at the begining and end of the path.

>Cannot found plug-in folder /usr/local/lib

Like in this print

maxim-smirnov, 2016-11-07:
> Cannot found plug-in folder /usr/local/lib
It's printing (*it):
`LOG(LogLevel::WARNINGS) << "Cannot found plug-in folder " << (*it);`

In my LOG call:
LOG(LogLevel::PLUGINS) << "For " << pluginDir << "\nIs existing: " << boost::filesystem::exists(pluginDir) << "\nIs it a directory: " << boost::filesystem::is_directory(pluginDir);

Liryna, 2016-11-07:
@maxim-smirnov what is the output of your log ?

maxim-smirnov, 2016-11-07:
It's defiantly problem with boost:
http://pastebin.com/4AQmU5h1

Liryna, 2016-11-07:
That's really surprising 😮 !
Do you think we should do something about it ? Check in another way ?
If you remove the line, does he load correctly the plugins ?

maxim-smirnov, 2016-11-07:
Removed if-else, scanning without exist checking:
```
2016-Nov-07 23:14:34.486812 - INFO: 	{79}	{getInstance}	{/home/user/liblogicalaccess/src/settings.cpp}:
New settings instance created.
2016-Nov-07 23:14:34.486843 - INFO: 	{26}	{dll_init_s}	{/home/user/liblogicalaccess/src/dynlibrary/dllinit.cpp}:
Process attached !
2016-Nov-07 23:14:34.486863 - PLUGIN: 	{248}	{scanPlugins}	{/home/user/liblogicalaccess/src/dynlibrary/librarymanager.cpp}:
Will scan 2 folders.
2016-Nov-07 23:14:34.486878 - PLUGIN: 	{252}	{scanPlugins}	{/home/user/liblogicalaccess/src/dynlibrary/librarymanager.cpp}:
For "/home/user/Projects/build-LibLigAccessTEST-Desktop-Debug"
Is existing: 0
Is it a directory: 0
2016-Nov-07 23:14:34.486891 - PLUGIN: 	{255}	{scanPlugins}	{/home/user/liblogicalaccess/src/dynlibrary/librarymanager.cpp}:
Scanning library folder /home/user/Projects/build-LibLigAccessTEST-Desktop-Debug ...
boost::filesystem::directory_iterator::construct: No such file or directory
2016-Nov-07 23:14:34.486960 - INFO: 	{31}	{~dll_init_s}	{/home/user/liblogicalaccess/src/dynlibrary/dllinit.cpp}:
Process detached !
Press <RETURN> to close this window...
```

Liryna, 2016-11-07:
😥 this is totally broken

Does even a boost sample behave the same ?

maxim-smirnov, 2016-11-07:
May be it's a bug in boost, cause running this:
```
#include <iostream>
#include <boost/filesystem.hpp>

int main(){
    boost::filesystem::path hostsPath("/etc/hosts");
    std::cout << boost::filesystem::exists(hostsPath) << '\n';
    return 0;
}
```
Returns:
`0`

Liryna, 2016-11-07:
😢 Ok well I hope this will be fixed by boost

maxim-smirnov, 2016-11-07:
But! Building this boost code with clang gives working program.
My be try to build your lib with clang?)

maxim-smirnov, 2016-11-07:
Ok, thank you for wasted time. I'll try to do something with this. 😄

maxim-smirnov, 2016-11-07:
Do not beat me, but I've compiled your lib with clang, now the output of the log is:
```
2016-Nov-08 00:51:56.959863 - INFO: 	{79}	{getInstance}	{/home/user/liblogicalaccess/src/settings.cpp}:
New settings instance created.
2016-Nov-08 00:51:56.959910 - INFO: 	{26}	{dll_init_s}	{/home/user/liblogicalaccess/src/dynlibrary/dllinit.cpp}:
Process attached !
/home/user/Projects/build-LibLigAccessTEST-Desktop-Debug/LibLigAccessTEST: symbol lookup error: /home/user/Projects/build-LibLigAccessTEST-Desktop-Debug/LibLigAccessTEST: undefined symbol: _ZN13logicalaccess14LibraryManager19getAvailableReadersEv

```

Liryna, 2016-11-07:
Where did you find this function 😮  `getAvailableReaders` ?

maxim-smirnov, 2016-11-07:
In LibraryManager class?)

Ok, sample now working. Plugins are loading. But here is little stump:
```
Please type the reader plug-in to use:
PCSC
1 readers on this system.
Please select index of the reader unit to use:
	0. VMware Virtual USB CCID 00 00
0
Waiting 15 seconds for a card insertion...
Time start : 1478557930
terminating with uncaught exception of type logicalaccess::LibLogicalAccessException: Failed to establish PCSC connection: PCSC error: 80100066. 
Press <RETURN> to close this window...
```
When I'm attaching any card.

maxim-smirnov, 2016-11-07:
So here is the full log after scanPlugins operation.
http://pastebin.com/WEyx3udV
I really believe that you'll help me, cause now i'm very close to the solution. I've dug toons of code, forums, bug reports, issues trackers, and now it's launching, and almost working.
❤️

Liryna, 2016-11-08:
Your card/VMware reader seems to not like RESET pcsc command. (SCARD_W_UNRESPONSIVE_CARD	0x80100066L	The smart card is not responding to a reset.) https://msdn.microsoft.com/en-us/library/ms936965.aspx

The exception happen here: https://github.com/islog/liblogicalaccess/blob/master/plugins/pluginsreaderproviders/pcsc/pcsc_connection.cpp#L25

I have no idea why this is happening :'( Maybe should "play" with the parameters of SCardConnect to see what `VMware Virtual USB CCID` accept.

maxim-smirnov, 2016-11-08:
I've connected reader with another interface. Now everything is fully working. Thank you for support!

closed 2016-11-08 by maxim-smirnov

Liryna, 2016-11-08:
No problem @maxim-smirnov and thanks for the feedback !