liblogicalaccess/liblogicalaccess #123: Problem with instalation this library

TomHyhlik, 2016-12-16:
Hi, I am using Linux Mint and I am a little new here.
When I am following instructions of adding the repository, after final comand apt-get update it returns this:
E: Malformed line 1 in source list /etc/apt/sources.list.d/liblogicalaccess.list (dist parse)
E: The list of sources could not be read.
Please hel me with that. Thanks.

Liryna, 2016-12-16:
Hi,

look liks the content of liblogicalaccess.list is incorrect. The script that you used is made for debian. You need to set a correct distrib in liblogicalaccess.list.

TomHyhlik, 2016-12-16:
Thanks for reply. This is content of my liblogicalaccess.list, but I don't really understand that. How should I modify it?

deb http://debian.islog.com/  main
deb http://debian.islog.com/  main
deb http://debian.islog.com/  main
deb http://debian.islog.com/  main
aaa
gnet
exit
afbbt\
deb http://debian.islog.com/  main
deb http://debian.islog.com/  main
deb http://debian.islog.com/  main

TomHyhlik, 2016-12-16:
Thanks for reply. I don't really understand it. How should I modify it?
deb http://debian.islog.com/ main

Liryna, 2016-12-17:
You have to use on of the repository name addpted to your Mint.
https://github.com/islog/liblogicalaccess/wiki/Install-LibLogicalAccess#debianubuntu-installation

It should be for exemple: deb http://debian.islog.com/ xenial main

But never tested on mint, you will probably need to rebuild yourself

TomHyhlik, 2016-12-17:
Finally the update worked but next installing command  
apt-get install liblogicalaccess 
reported 
E: Unable to locate package liblogicalaccess
so there must be something wrong with the repository.

Liryna, 2016-12-17:
@TomHyhlik Our CI use the repository with out issue. Try to change the distribution to stable.

Maxhy, 2016-12-22:
Looks like an issue related to your distrib/environment. I believe this is something that have to be fixed on your side, not something we can help with. Feel free to comment here about the resolution.

closed 2016-12-22 by Maxhy

TomHyhlik, 2017-01-12:
I reinstalled my os to ubuntu 16.04 LTS and than it woeks, sorry for annoying...