liblogicalaccess/liblogicalaccess #125: Assertion failed: provider. Cannot get PCSC provider

FeitianSmartcardReader, 2017-01-09:
Hi 

I just want test this lib work with PC/SC reader, my reader is Feitian R502 which is PC/SC reader, the testpcsc works and also the PC/SC API works, does anyone know why give this error? 

ben-virtual-machine ben # /usr/local/sbin/pcscd -adf
00000000 pcscdaemon.c:346:main() pcscd set to foreground with debug send to stdout
00000083 configfile.l:282:DBGetReaderListDir() Parsing conf directory: /usr/local/etc/reader.conf.d
00000020 configfile.l:319:DBGetReaderListDir() Skipping non regular file: ..
00000020 configfile.l:319:DBGetReaderListDir() Skipping non regular file: .
00000011 pcscdaemon.c:655:main() pcsc-lite 1.8.18 daemon ready.
00001296 hotplug_libusb.c:536:HPAddHotPluggable() Adding USB device: 2:6:0
02602721 hotplug_libusb.c:590:HPAddHotPluggable() libusb_get_string_descriptor_ascii failed: -7
00000025 readerfactory.c:1079:RFInitializeReader() Attempting startup of Feitian 502-CL 00 00 using /usr/local/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000100 readerfactory.c:954:RFBindFunctions() Loading IFD Handler 3.0
00000018 ifdhandler.c:1953:init_driver() Driver version: 1.4.24
00000415 ifdhandler.c:1970:init_driver() LogLevel: 0x0003
00000005 ifdhandler.c:1981:init_driver() DriverOptions: 0x0000
00000111 ifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 0, device: usb:096e/0608:libusb-1.0:2:6:0
00000016 ccid_usb.c:287:OpenUSBByName() Using: /usr/local/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000374 ccid_usb.c:305:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000005 ccid_usb.c:306:OpenUSBByName() ifdProductString: Generic CCID driver
00000003 ccid_usb.c:307:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00000552 ccid_usb.c:621:OpenUSBByName() Found Vendor/Product: 096E/0608 (Feitian 502-CL)
00000008 ccid_usb.c:623:OpenUSBByName() Using USB bus/device: 2/6
01999802 ccid_usb.c:1204:ControlUSB() control failed (2/6): -9 LIBUSB_ERROR_PIPE
00000036 ccid_usb.c:1135:get_data_rates() IFD does not support GET_DATA_RATES request: -9
00121790 ifdhandler.c:379:IFDHGetCapabilities() tag: 0xFB3, usb:096e/0608:libusb-1.0:2:6:0 (lun: 0)
00000026 readerfactory.c:395:RFAddReader() Using the reader polling thread
00003974 ifdhandler.c:379:IFDHGetCapabilities() tag: 0xFAE, usb:096e/0608:libusb-1.0:2:6:0 (lun: 0)
00000018 ifdhandler.c:470:IFDHGetCapabilities() Reader supports 1 slot(s)
00000173 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00000014 hotplug_libusb.c:440:HPEstablishUSBNotifications() Driver ifd-ccid.bundle does not support IFD_GENERATE_HOTPLUG. Using active polling instead.
00000004 hotplug_libusb.c:449:HPEstablishUSBNotifications() Polling forced every 1 second(s)
00008758 ifdhandler.c:1146:IFDHPowerICC() action: PowerUp, usb:096e/0608:libusb-1.0:2:6:0 (lun: 0)
00001635 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00010269 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00013787 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00010906 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00010131 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00010177 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00010228 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00010652 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00011258 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00017132 readerfactory.c:1420:RFWaitForReaderInit() Waiting init for reader: Feitian 502-CL 00 00
00005891 eventhandler.c:286:EHStatusHandlerThread() powerState: POWER_STATE_POWERED
00000026 Card ATR: 3B 81 80 01 80 80

Liryna, 2017-01-09:
Hi @FeitianSmartcardReader ,

What is you error ?

FeitianSmartcardReader, 2017-01-09:
[image]
here is screenshot

Liryna, 2017-01-09:
You have to copie https://github.com/islog/liblogicalaccess/blob/master/liblogicalaccess.config in the same folder as the test and change the path https://github.com/islog/liblogicalaccess/blob/master/liblogicalaccess.config#L29 to the location of libpcscreaders.so that you build.

Could it be possible to have an Feitian smartcard reader to do the test ? :heart:

FeitianSmartcardReader, 2017-01-09:
Yes, sure. send me mail with your address and phone, I will ship to you.

FeitianSmartcardReader, 2017-01-09:
works now. Thank you very much, ;-)

closed 2017-01-09 by FeitianSmartcardReader