liblogicalaccess/liblogicalaccess #127: debian : apt-get unable to locate package

neywen, 2017-02-13:
Hello,
in a Debian virtual machine, I'm trying to install the liblogicalaccess, using apt-get, as mentioned in the "installation" pages. 

```
$ sudo apt-get install liblogicalaccess-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package liblogicalaccess-dev

```

should I download the repository and compile the lib? Or is there any trick I don't know about ?
Thank you for any help,
l.

Liryna, 2017-02-13:
Hi @neywen ,

Did you add the repository and apt-get update ?
https://github.com/islog/liblogicalaccess/wiki/Install-LibLogicalAccess#debianubuntu-installation

neywen, 2017-02-13:
Hi, thank you for your message.
I've done it, yes, but I've done it the wrong way.

Now I think I get it right : 

```
export REPOSITORY_NAME=liblogicalaccess
echo "deb http://debian.islog.com/ ${REPOSITORY_NAME} main" | sudo tee -a /etc/apt/sources.list.d/liblogicalaccess.list
sudo curl http://debian.islog.com/debian-islog-com-keyring.key | sudo apt-key add -
sudo apt-get update

```

But I get 404 errors with the following output : 

```
$ sudo apt-get update
Ign http://debian.islog.com liblogicalaccess Release.gpg
Ign http://debian.islog.com liblogicalaccess Release                        
Hit http://security.debian.org wheezy/updates Release.gpg                                                               
Hit http://httpredir.debian.org wheezy Release.gpg                                                                      
Hit http://security.debian.org wheezy/updates Release
Hit http://httpredir.debian.org wheezy-updates Release.gpg           
Hit http://security.debian.org wheezy/updates/main Sources            
Hit http://httpredir.debian.org wheezy Release                       
Hit http://security.debian.org wheezy/updates/contrib Sources                           
Hit http://httpredir.debian.org wheezy-updates Release                                                        
Hit http://security.debian.org wheezy/updates/non-free Sources                                                
Err http://debian.islog.com liblogicalaccess/main amd64 Packages      
  404  Not Found
Ign http://debian.islog.com liblogicalaccess/main Translation-en_US                                       
Hit http://security.debian.org wheezy/updates/main amd64 Packages                           
Hit http://httpredir.debian.org wheezy/main Sources                                                             
Ign http://debian.islog.com liblogicalaccess/main Translation-en                                                
Hit http://security.debian.org wheezy/updates/contrib amd64 Packages                                          
Hit http://httpredir.debian.org wheezy/non-free Sources                                 
Hit http://security.debian.org wheezy/updates/non-free amd64 Packages
Hit http://httpredir.debian.org wheezy/contrib Sources            
Hit http://security.debian.org wheezy/updates/contrib Translation-en
Hit http://security.debian.org wheezy/updates/main Translation-en
Hit http://httpredir.debian.org wheezy/main amd64 Packages        
Hit http://security.debian.org wheezy/updates/non-free Translation-en
Hit http://httpredir.debian.org wheezy/non-free amd64 Packages    
Hit http://httpredir.debian.org wheezy/contrib amd64 Packages     
Hit http://httpredir.debian.org wheezy/contrib Translation-en
Hit http://httpredir.debian.org wheezy/main Translation-en
Hit http://httpredir.debian.org wheezy/non-free Translation-en
Hit http://httpredir.debian.org wheezy-updates/main Sources
Hit http://httpredir.debian.org wheezy-updates/contrib Sources
Hit http://httpredir.debian.org wheezy-updates/non-free Sources
Hit http://httpredir.debian.org wheezy-updates/main amd64 Packages/DiffIndex
Hit http://httpredir.debian.org wheezy-updates/contrib amd64 Packages     
Hit http://httpredir.debian.org wheezy-updates/non-free amd64 Packages/DiffIndex
Hit http://httpredir.debian.org wheezy-updates/contrib Translation-en
Hit http://httpredir.debian.org wheezy-updates/main Translation-en/DiffIndex
Hit http://httpredir.debian.org wheezy-updates/non-free Translation-en/DiffIndex
W: Failed to fetch http://debian.islog.com/dists/liblogicalaccess/main/binary-amd64/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

does it ring a bell ?

Liryna, 2017-02-13:
>export REPOSITORY_NAME=liblogicalaccess

should be

>export REPOSITORY_NAME=stable

Since you use stable (wheezy)

neywen, 2017-02-13:
ok, that's what the documentation says, I misunderstood. 

changes are made, but getting the same results : 

```
$ echo ${REPOSITORY_NAME}
stable

$ echo "deb http://debian.islog.com/ ${REPOSITORY_NAME} main" | sudo tee -a /etc/apt/sources.list.d/liblogicalaccess.list
deb http://debian.islog.com/ stable main

$ sudo curl http://debian.islog.com/debian-islog-com-keyring.key | sudo apt-key add -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1707  100  1707    0     0  51966      0 --:--:-- --:--:-- --:--:--  185k
OK

$ sudo apt-get update
Ign http://debian.islog.com liblogicalaccess Release.gpg
Hit http://security.debian.org wheezy/updates Release.gpg           
Hit http://debian.islog.com stable Release.gpg                      
Ign http://debian.islog.com liblogicalaccess Release                 
Hit http://security.debian.org wheezy/updates Release                                    
Hit http://debian.islog.com stable Release                                                                      
Hit http://security.debian.org wheezy/updates/main Sources                                                                       
Hit http://httpredir.debian.org wheezy Release.gpg                                                        
Hit http://security.debian.org wheezy/updates/contrib Sources         
Hit http://httpredir.debian.org wheezy-updates Release.gpg            
Hit http://debian.islog.com stable/main amd64 Packages                
Hit http://security.debian.org wheezy/updates/non-free Sources                           
Hit http://httpredir.debian.org wheezy Release                                      
Hit http://security.debian.org wheezy/updates/main amd64 Packages                           
Hit http://httpredir.debian.org wheezy-updates Release                                                          
Hit http://security.debian.org wheezy/updates/contrib amd64 Packages                        
Hit http://security.debian.org wheezy/updates/non-free amd64 Packages                       
Hit http://httpredir.debian.org wheezy/main Sources                                  
Hit http://security.debian.org wheezy/updates/contrib Translation-en                    
Hit http://httpredir.debian.org wheezy/non-free Sources                                 
Hit http://security.debian.org wheezy/updates/main Translation-en                       
Hit http://security.debian.org wheezy/updates/non-free Translation-en                   
Hit http://httpredir.debian.org wheezy/contrib Sources                                  
Hit http://httpredir.debian.org wheezy/main amd64 Packages                              
Hit http://httpredir.debian.org wheezy/non-free amd64 Packages    
Hit http://httpredir.debian.org wheezy/contrib amd64 Packages     
Hit http://httpredir.debian.org wheezy/contrib Translation-en     
Hit http://httpredir.debian.org wheezy/main Translation-en        
Hit http://httpredir.debian.org wheezy/non-free Translation-en    
Hit http://httpredir.debian.org wheezy-updates/main Sources       
Err http://debian.islog.com liblogicalaccess/main amd64 Packages  
  404  Not Found
Ign http://debian.islog.com liblogicalaccess/main Translation-en_US
Hit http://httpredir.debian.org wheezy-updates/contrib Sources    
Ign http://debian.islog.com liblogicalaccess/main Translation-en  
Hit http://httpredir.debian.org wheezy-updates/non-free Sources                            
Ign http://debian.islog.com stable/main Translation-en_US                                       
Ign http://debian.islog.com stable/main Translation-en                                   
Hit http://httpredir.debian.org wheezy-updates/main amd64 Packages/DiffIndex             
Hit http://httpredir.debian.org wheezy-updates/contrib amd64 Packages
Hit http://httpredir.debian.org wheezy-updates/non-free amd64 Packages/DiffIndex
Hit http://httpredir.debian.org wheezy-updates/contrib Translation-en
Hit http://httpredir.debian.org wheezy-updates/main Translation-en/DiffIndex
Hit http://httpredir.debian.org wheezy-updates/non-free Translation-en/DiffIndex
W: Failed to fetch http://debian.islog.com/dists/liblogicalaccess/main/binary-amd64/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Liryna, 2017-02-13:
Try to remove/purge /etc/apt/sources.list.d/liblogicalaccess.list and exec the commands again.

neywen, 2017-02-13:
ok, that's fixed the problem.
Thank you a lot for your help Liryna !
l.

closed 2017-02-13 by neywen

xinthose, 2017-12-19:
```
echo "deb http://debian.islog.com/ stable main" | sudo tee -a /etc/apt/sources.list.d/liblogicalaccess.list
sudo curl http://debian.islog.com/debian-islog-com-keyring.key | sudo apt-key add -
sudo apt-get update
sudo apt-get install liblogicalaccess
```