liblogicalaccess/liblogicalaccess #133: Cannot store data on card

unregistered436, 2017-05-06:
Hello,

I have a Dangerous Things implant that I'm trying to use with ISLOG. On first login I can enroll a tag to unlock my current session, but I cannot store credentials for ongoing authentication.

The chip in this implant is an NXP NTAG216. I used the Dangerous Things app to secure the tag and lock the CC (see this post for a good explanation https://forum.dangerousthings.com/t/dnfc-and-password-protection/532/2)

So, the tag is now write-only.  I can put any number of NDEF records on it using NFC tagwriter. For some reason though ISLOG says it "cannot store data on card" when I try to enroll it. Is it expecting to completely unlock the chip before writing to it?

Thanks,
Unreg

Maxhy, 2017-05-10:
Thanks for the report. This is because current NFC Tag implementation assume that we are the only and unique NFC record. We don't check read/write attributes neither lock bytes, whereas to support your case we should. We are aware about the current implementation limitation but no GitHub issue existed, thanks for creating one ;)

Maxhy, 2019-06-21:
Lock bytes are now supported thanks to @Holocrons.

It is going to be supported on next ISLOG Logon NFC Community version (September 2019). As ISLOG Logon product is outside LibLogicalAccess library scope, I'm going to close this issue here for now.

closed 2019-06-21 by Maxhy

unregistered436, 2019-12-09:
Was the September 2019 release published? The only version I can download from http://www.islog.com/en/software/islog-logon-nfc-community/ is 4.8.0531. This is the same version I downloaded back in February 2019.

Maxhy, 2019-12-10:
It wasn't released yet. It has been delayed but still on track... new target date is end of December for this product update.