liblogicalaccess/liblogicalaccess #134: failed in maven build

bondhan, 2017-05-16:
Hi,

Is this project still active? I found difficulties to build the source code and stumbled on maven error below:

[ERROR] Failed to execute goal on project liblogicalaccess: Could not resolve dependencies for project eu.islog.lib.readers:liblogicalaccess:pom:1.85.0: The following artifacts could not be resolved: lib.portable:boost:zip:1.57-light2, lib.crypto:openssl:zip:1.0.0b.2, lib.readers:scardsyn:zip:2.0.0.0, lib.readers:rfideas:zip:7.1.5: Could not find artifact lib.portable:boost:zip:1.57-light2 in central (https://repo.maven.apache.org/maven2) -> [Help 1]

Do you have any advice?

Liryna, 2017-05-16:
Hi @bondhan ,

You need to add islog maven repository to your settings
https://github.com/islog/liblogicalaccess/wiki/Build-Liblogicalaccess#maven-installation

Maxhy, 2017-05-22:
Closing as invalid.

closed 2017-05-22 by Maxhy