liblogicalaccess/liblogicalaccess #147: Issue connecting to ACR1252.

xinthose, 2017-12-20:
Hello,

   I am following your Quick How To (C++) guide and am getting the following error:
`The ATR length must be at least 2 bytes long (TS + T0).`
after this cout:
`Waiting 15 seconds for a card insertion...`
with the following code:
```
#include <logicalaccess/dynlibrary/librarymanager.hpp>
#include <logicalaccess/readerproviders/readerconfiguration.hpp>
#include <logicalaccess/cards/chip.hpp>
#include <logicalaccess/bufferhelper.hpp>
#include <logicalaccess/myexception.hpp>

#include <iostream>
#include <string>
#include <vector>

int main (void)
{
	try
	{
		// Reader configuration object to store reader provider and reader unit selection.
		std::shared_ptr<logicalaccess::ReaderConfiguration> readerConfig(new logicalaccess::ReaderConfiguration());
		 
		// Set PCSC ReaderProvider by calling the Library Manager which will load the function from the corresponding plug-in
		readerConfig->setReaderProvider(logicalaccess::LibraryManager::getInstance()->getReaderProvider("PCSC"));

		// Create the default reader unit. On PC/SC, we will listen on all readers.
		readerConfig->setReaderUnit(readerConfig->getReaderProvider()->createReaderUnit());
		if (readerConfig->getReaderProvider()->getRPType() == "PCSC" && readerConfig->getReaderProvider()->getReaderList().size() == 0)
		{
			std::cerr << "No readers on this system." << std::endl;
			return EXIT_FAILURE;
		}
		std::cout << readerConfig->getReaderProvider()->getReaderList().size() << " readers on this system." << std::endl;

		readerConfig->getReaderUnit()->connectToReader();

		// get card information
		std::cout << "Waiting 15 seconds for a card insertion..." << std::endl;
		if (readerConfig->getReaderUnit()->waitInsertion(15000))
		{
			if (readerConfig->getReaderUnit()->connect())
			{
				 // Display few chip information

				 std::cout << "Card inserted on reader \"" << readerConfig->getReaderUnit()->getConnectedName() << "\"." << std::endl;

				 std::shared_ptr<logicalaccess::Chip> chip = readerConfig->getReaderUnit()->getSingleChip();
				 std::cout << "Card type: " << chip->getCardType() << std::endl;

				 std::vector<unsigned char> csn = readerConfig->getReaderUnit()->getNumber(chip);
				 std::cout << "Card unique manufacturer number : " << logicalaccess::BufferHelper::getHex(csn) << std::endl;

				 // DO CHIP COMMAND HERE
				 // DO CHIP COMMAND HERE
				 // DO CHIP COMMAND HERE

				readerConfig->getReaderUnit()->disconnect();
			}

			std::cout << "Logical automatic card removal in 15 seconds..." << std::endl;
			if (!readerConfig->getReaderUnit()->waitRemoval(15000))
			{
				 std::cerr << "Card removal forced." << std::endl;
			}

			std::cout << "Card removed." << std::endl;
		}
		else
		{
			std::cout << "No card inserted." << std::endl;
		}
		
		// Disconnect
		readerConfig->getReaderUnit()->disconnectFromReader();		
	}
	catch (logicalaccess::LibLogicalAccessException &ex)
	{
		std::cout << "LibLogicalAccess exception: " << ex.what() << std::endl;
		return EXIT_FAILURE;
	}
	catch (const std::exception &e)
	{
		std::cerr << "e.what() = " << e.what() << std::endl;
		return EXIT_FAILURE;
	}
	
	return EXIT_SUCCESS;
}
```
- I am using this reader: [ACR1252](https://www.amazon.com/ACR1252U-USB-NFC-Reader-III/dp/B01KIKBYAG/ref=sr_1_1?ie=UTF8&qid=1513782405&sr=8-1&keywords=acr1252u )
- I get this message: "2 readers on this system." before I get the above error.  I have only the ACR1252 plugged in when I run my code.  I get "No readers on this system." with it unplugged.

I realize that I am probably using the wrong reader of the 2 from this one reader.  How do I select a different reader?  What am I doing wrong?  Thank you for this excellent library.

Liryna, 2017-12-20:
Hi @xinthose ,

You need to select the correct reader. You are probably using the contact reader here.
See https://github.com/islog/liblogicalaccess/blob/master/samples/basic/basic.cpp#L62

xinthose, 2017-12-20:
@Liryna Thank you.  I figured out that I should be using this code:
```
const logicalaccess::ReaderList readers = readerConfig->getReaderProvider()->getReaderList();
readerConfig->setReaderUnit(readers.at(0));
```
instead of this line:
`readerConfig->setReaderUnit(readerConfig->getReaderProvider()->createReaderUnit());`
 This example helped me to figure this out:  [liblogicalaccess/samples/basic/basic.cpp ](https://github.com/islog/liblogicalaccess/blob/master/samples/basic/basic.cpp)

closed 2017-12-20 by xinthose