liblogicalaccess/liblogicalaccess #148: Mifare Ultralight writePage

xinthose, 2017-12-21:
https://github.com/islog/liblogicalaccess/blob/34c3335d6c23d22dfc932e64116928dc8f80ac87/plugins/pluginscards/mifareultralight/mifareultralightcommands.hpp#L68

        /**
         * \brief Write a whole page.
         * \param sector The page number, from 0 to 15.
         * \param buf A buffer to from which to copy the data.
         * \param buflen The length of buffer. Must be at least 4 bytes long or the call will fail.
         * \return The number of bytes written, or a negative value on error.
         */

There is no return from the void function.

Liryna, 2017-12-22:
Hi @xinthose ,,

Thank you for the report.
I have rename the issue because this is not only linked to ultralight. We shouled review all the documentation of LLA.

xinthose, 2017-12-23:
@Liryna You're welcome ma'am.  I have seen the same thing you have said: inconsistent documentation (or incorrect functions).