liblogicalaccess/liblogicalaccess #159: Make error while building on raspberry pi

pwierzchon, 2018-06-27:
Hi, I wanted to build the library in order to use it on my RPi. But while executing make I get the following communicate:
```
[  4%] Building CXX object plugins/cryptolib/CMakeFiles/logicalaccess-cryptolib.dir/src/evp_pkey.cpp.o
/home/pi/liblogicalaccess-1.84.0/plugins/cryptolib/src/evp_pkey.cpp: In member function ‘int logicalaccess::openssl::EVPPKey::type() const’:
/home/pi/liblogicalaccess-1.84.0/plugins/cryptolib/src/evp_pkey.cpp:32:46: error: invalid use of incomplete type ‘struct evp_pkey_st’
             return EVP_PKEY_type(d_pkey.get()->type);
                                              ^~
In file included from /usr/include/openssl/evp.h:14:0,
                 from /home/pi/liblogicalaccess-1.84.0/plugins/cryptolib/include/logicalaccess/crypto/evp_pkey.hpp:16,
                 from /home/pi/liblogicalaccess-1.84.0/plugins/cryptolib/src/evp_pkey.cpp:7:
/usr/include/openssl/ossl_typ.h:93:16: note: forward declaration of ‘struct evp_pkey_st’
 typedef struct evp_pkey_st EVP_PKEY;
                ^~~~~~~~~~~
plugins/cryptolib/CMakeFiles/logicalaccess-cryptolib.dir/build.make:518: recipe for target 'plugins/cryptolib/CMakeFiles/logicalaccess-cryptolib.dir/src/evp_pkey.cpp.o' failed
make[2]: *** [plugins/cryptolib/CMakeFiles/logicalaccess-cryptolib.dir/src/evp_pkey.cpp.o] Error 1
CMakeFiles/Makefile2:122: recipe for target 'plugins/cryptolib/CMakeFiles/logicalaccess-cryptolib.dir/all' failed
make[1]: *** [plugins/cryptolib/CMakeFiles/logicalaccess-cryptolib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```
Do you have any ideas how can I build it on RPi? Is it even possible?

Liryna, 2018-06-27:
@pwierzchon Hi,

You seems to have the wrong OpenSSL version installed or cmake cannot target the correct include directory.

pwierzchon, 2018-07-05:
Thanks. Is there a way to know what version of openssl should I use to build this?

Liryna, 2018-07-05:
Depends the branch that you try to build:
See https://github.com/islog/liblogicalaccess/blob/master/pom.xml#L779
https://github.com/islog/liblogicalaccess/blob/develop/pom.xml#L780

Liryna, 2019-05-17:
Closing for inactivity.

closed 2019-05-17 by Liryna