liblogicalaccess/liblogicalaccess #161: Omnikey 5427 - get card tag in human readable form?

xvoidee, 2018-10-14:
Hello,

I'm using HID Omnikey 5427 under Ubuntu 14.04.5 amd64.
Everything is fine except that I am not able to get card ID (it is 6 digits written on the card - 658399).
Card class is HID iCLASS Seos Px H8Y.

This is my sandbox code (modified examples). Which part of API should I use in order to get this ID value?
When I run code I always see that card serial number is changing:
Card Serial Number : 08E8D0A1
Card Serial Number : 08DDEA85
Card Serial Number : 08FFFFFF

Card ATR is
3B 80 80 01 01

Any info is welcome. Thanks!

`/**
 * \file main.cpp
 * \author Maxime C. <maxime-dev@islog.com>
 * \brief Main file for the basic sample of the liblogicalaccess.
 */

#include "logicalaccess/dynlibrary/idynlibrary.hpp"
#include "logicalaccess/dynlibrary/librarymanager.hpp"
#include "logicalaccess/readerproviders/readerconfiguration.hpp"
#include "logicalaccess/services/storage/storagecardservice.hpp"
#include "logicalaccess/services/accesscontrol/accesscontrolcardservice.hpp"
#include "logicalaccess/services/accesscontrol/formats/wiegand26format.hpp"
#include "logicalaccess/services/nfctag/nfctagcardservice.hpp"
#include "logicalaccess/cards/locationnode.hpp"

#include <pcsc/readers/omnikey5427readerunitconfiguration.hpp>
#include <pcsc/readers/omnikeyxx22readerunit.hpp>
#include <pcsc/readers/omnikeyxx23readerunit.hpp>
#include <pcsc/readers/omnikeyxx25readerunit.hpp>
#include <pcsc/readers/acsacrreaderunit.hpp>

#include <iostream>
#include <string>
#include <iomanip>
#include <sstream>
#include <stdlib.h>

/**
 * \brief The application entry point.
 * \param argc The arguments count.
 * \param argv The arguments.
 */
int main(int , char**)
{
    try
    {
        std::vector<std::string> readerList = logicalaccess::LibraryManager::getInstance()->getAvailableReaders();
        std::cout << "Available reader plug-ins ("<< readerList.size() <<"):" << std::endl;
        for (std::vector<std::string>::iterator it = readerList.begin(); it != readerList.end(); ++it)
        {
            std::cout << "\t" << (*it) << std::endl;
        }

        std::vector<std::string> cardList = logicalaccess::LibraryManager::getInstance()->getAvailableCards();
        std::cout << "Available card plug-ins ("<< cardList.size() <<"):" << std::endl;
        for (std::vector<std::string>::iterator it = cardList.begin(); it != cardList.end(); ++it)
        {
            std::cout << "\t" << (*it) << std::endl;
        }

        // Reader configuration object to store reader provider and reader unit selection.
        std::shared_ptr<logicalaccess::ReaderConfiguration> readerConfig(new logicalaccess::ReaderConfiguration());

[... 120 lines cut ...]
                {
                    std::cout << "Error: cannot connect to the card." << std::endl;
                }

                std::cout << "Logical automatic card removal in 15 seconds..." << std::endl;

                if (!readerConfig->getReaderUnit()->waitRemoval(15000))
                {
                    std::cerr << "Card removal forced." << std::endl;
                }

                std::cout << "Card removed." << std::endl;
            } else
            {
                std::cout << "No card inserted." << std::endl;
            }
        }
    }
    catch (std::exception& ex)
    {
        std::cout << "Exception: " << ex.what() << std::endl;
    }

    return EXIT_SUCCESS;
}`

xvoidee, 2018-10-14:
#include "logicalaccess/dynlibrary/idynlibrary.hpp"
#include "logicalaccess/dynlibrary/librarymanager.hpp"
#include "logicalaccess/readerproviders/readerconfiguration.hpp"
#include "logicalaccess/services/storage/storagecardservice.hpp"
#include "logicalaccess/services/accesscontrol/accesscontrolcardservice.hpp"
#include "logicalaccess/services/accesscontrol/formats/wiegand26format.hpp"
#include "logicalaccess/services/nfctag/nfctagcardservice.hpp"
#include "logicalaccess/cards/locationnode.hpp"

#include <pcsc/readers/omnikey5427readerunitconfiguration.hpp>
#include <pcsc/readers/omnikeyxx22readerunit.hpp>
#include <pcsc/readers/omnikeyxx23readerunit.hpp>
#include <pcsc/readers/omnikeyxx25readerunit.hpp>
#include <pcsc/readers/acsacrreaderunit.hpp>

#include <iostream>
#include <string>
#include <iomanip>
#include <sstream>
#include <stdlib.h>

int main(int , char**)
{
    try
    {
        std::vector<std::string> readerList = logicalaccess::LibraryManager::getInstance()->getAvailableReaders();
        std::cout << "Available reader plug-ins ("<< readerList.size() <<"):" << std::endl;
        for (std::vector<std::string>::iterator it = readerList.begin(); it != readerList.end(); ++it)
        {
            std::cout << "\t" << (*it) << std::endl;
        }

        std::vector<std::string> cardList = logicalaccess::LibraryManager::getInstance()->getAvailableCards();
        std::cout << "Available card plug-ins ("<< cardList.size() <<"):" << std::endl;
        for (std::vector<std::string>::iterator it = cardList.begin(); it != cardList.end(); ++it)
        {
            std::cout << "\t" << (*it) << std::endl;
        }

        // Reader configuration object to store reader provider and reader unit selection.
        std::shared_ptr<logicalaccess::ReaderConfiguration> readerConfig(new logicalaccess::ReaderConfiguration());

        // PC/SC
        std::string rpstr = "PCSC";
        std::cout << "Please type the reader plug-in to use:" << std::endl;
        //std::cin >> rpstr;

        readerConfig->setReaderProvider(logicalaccess::LibraryManager::getInstance()->getReaderProvider(rpstr));

        if (readerConfig->getReaderProvider()->getRPType() == "PCSC" && readerConfig->getReaderProvider()->getReaderList().size() == 0)
        {
            std::cerr << "No readers on this system." << std::endl;
            return EXIT_FAILURE;
        }
        std::cout << readerConfig->getReaderProvider()->getReaderList().size() << " readers on this system." << std::endl;

        // List available reader units
        int ruindex = 0;
        const logicalaccess::ReaderList readers = readerConfig->getReaderProvider()->getReaderList();
        std::cout << "Please select index of the reader unit to use:" << std::endl;
        for (int i = 0; i < readers.size(); ++i)
        {
            std::cout << "\t" << i << ". " << readers.at(i)->getName() << std::endl;
        }
        //std::cin >> ruindex;

        if (ruindex < 0 || ruindex >= readers.size())
        {
            std::cerr << "Selected reader unit outside range." << std::endl;
            return EXIT_FAILURE;
[... 92 lines cut ...]
                {
                    std::cout << "Error: cannot connect to the card." << std::endl;
                }

                std::cout << "Logical automatic card removal in 15 seconds..." << std::endl;

                if (!readerConfig->getReaderUnit()->waitRemoval(15000))
                {
                    std::cerr << "Card removal forced." << std::endl;
                }

                std::cout << "Card removed." << std::endl;
            } else
            {
                std::cout << "No card inserted." << std::endl;
            }
        }
    }
    catch (std::exception& ex)
    {
        std::cout << "Exception: " << ex.what() << std::endl;
    }

    return EXIT_SUCCESS;
}

Liryna, 2018-10-15:
Hi @xvoidee ,

This card is currently not supported in the public repository unfortunately for NDA reason.

xvoidee, 2018-10-15:
Thank you @Liryna,

How can we use private repository? It is mentioned in wiki, that iclass cards are available on private repo. Is it paid service?

Liryna, 2018-10-23:
Hi @xvoidee ,

Yes, it is a paid service that we can only start to propose for companies regarding the needs. If that’s the case, could you send a mail to contact@islog.com and describe your usage.

Liryna, 2019-05-17:
Closing for inactivity.

closed 2019-05-17 by Liryna