liblogicalaccess/liblogicalaccess #166: Segmentation Fault

alex-dranoel, 2019-05-17:
Hello, first of all congratulation for this library, it looks very helpful. 

However, I quickly ended with a segmentation fault while following basic examples.

After successful installation on ubuntu 18.04 via
`sudo apt-get install liblogicalaccess-dev`
I could easily build and execute the provided basic.cpp example via the command:
`g++ basic.cpp -std=c++11 -W -Wall -Wno-unused-parameter -llogicalaccess -DUNIX`

And output was as expected:
```
Waiting 15 seconds for a card insertion...
Time start : 1558073608
Card inserted on reader "SpringCard CSB6 Ultimate (DD6BBF5C) 00 00".
Card type: DESFireEV1
Card Serial Number : 042379B1B41E80
Root Location Node : Mifare DESFire EV1
Complete chip list:
	042379B1B41E80
```

I then tried to select the PICC application of my DESFire EV1 card by adding the following (and the header `#include <desfire/desfireev1chip.hpp>`):

```
std::shared_ptr<logicalaccess::DESFireEV1Chip> desfireev1Chip = std::dynamic_pointer_cast<logicalaccess::DESFireEV1Chip>(chip);

if (desfireev1Chip)
{
    std::cout << "This is indeed a DESFire EV1 chip." << std::endl;
    std::shared_ptr<logicalaccess::DESFireEV1Commands> desfireev1cmd = desfireev1Chip->getDESFireEV1Commands();

    desfireev1cmd->selectApplication(0x000000);
    std::cout << "Application 0x000000 selected successfully." << std::endl;
}
else
{
    std::cout << "Not a DESFire EV1 chip." << std::endl;
}

```

I could compile it with:
`g++ basic.cpp -I/usr/include/logicalaccess/plugins/cards/ -L/usr/lib/logicalaccess/ -std=c++11 -W -Wall -Wno-unused-parameter -llogicalaccess -ldesfirecards -DUNIX`

But during execution, the line `desfireev1cmd->selectApplication(0x000000);` gives a Segmentation fault.  
```
Waiting 15 seconds for a card insertion...
Time start : 1558073715
Card inserted on reader "SpringCard CSB6 Ultimate (DD6BBF5C) 00 00".
Card type: DESFireEV1
Card Serial Number : 042379B1B41E80
Root Location Node : Mifare DESFire EV1
Complete chip list:
	042379B1B41E80
This is indeed a DESFire EV1 chip.
Segmentation fault (core dumped)
```

Any idea about what's happening ? Am I missing something ? 
Thanks in advance.

Liryna, 2019-05-17:
Hi @alex-dranoel ,

Could you try to rebuild using the develop branch ? Maybe there is an issue with the debian packages.

alex-dranoel, 2019-05-18:
Hello, there is no issue, it was my mistake. The desfire ev1 selecetApplication command does not directly takes an integer but needs a Location. And the 0x00000 was interpreted as a null location I guess. So it compiled but crashed at run time.

Might be a good place for an explicit constructor maybe.

Thanks for the quick answer anyway. Exploration continues on my side :)

closed 2019-05-18 by alex-dranoel