liblogicalaccess/liblogicalaccess #181: CMake Error

minimalisticMe, 2019-11-16:
Hey, I just downloaded the master-branch of liblogicalaccess on Windows 10 and try to create the Solutionfile for VS2019 `v16.3.9` and CMake `v3.16.0`.

First downloaded dependencies via `cmd` , calling `conan install` and it was a success after changing the boost library to `1.71.0`.
However when trying to `Configure` the project via CMake afterwards, I get a couple of errors:

```
CMake Error at CMakeLists.txt:126 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at plugins/logicalaccess/plugins/iks/CMakeLists.txt:20 (SET_TARGET_PROPERTIES):
  SET_TARGET_PROPERTIES called with incorrect number of arguments.

CMake Error at plugins/logicalaccess/plugins/cards/em4102/CMakeLists.txt:22 (SET_TARGET_PROPERTIES):
  SET_TARGET_PROPERTIES called with incorrect number of arguments.

CMake Error at plugins/logicalaccess/plugins/cards/em4135/CMakeLists.txt:20 (SET_TARGET_PROPERTIES):
  SET_TARGET_PROPERTIES called with incorrect number of arguments.

...
```
How can I fix this inconvenience?

minimalisticMe, 2019-11-16:
Closing - I was using `conan` wrong

closed 2019-11-16 by minimalisticMe

roelofd, 2020-05-18:
I have the exact same problem while trying to build the library. What was the fix for this?