liblogicalaccess/liblogicalaccess #186: Program run error: "throwing an instance of 'std::logic_error'"

roelofd, 2020-05-27:
I have installed the 2.2.1 version of the library on Kubuntu 18.04 with CMake 3.17.1 and Conan 1.25.0. 
I tried to follow the [C++ howto](https://github.com/islog/liblogicalaccess/wiki/How-to-Use-Liblogicalaccess) of the liblogicalaccess library and on the second line of code, when running the program it gives the following error:  
```
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
./run.sh: line 9:  6781 Aborted                 (core dumped) bin/./github_console
```

**main.cpp**
```
// my first program in C++
#include <iostream>

#include <logicalaccess/dynlibrary/librarymanager.hpp>
#include <logicalaccess/readerproviders/readerconfiguration.hpp>
#include <logicalaccess/cards/chip.hpp>

int main()
{
    std::cout << "Hello World!\n";
  
    // Reader configuration object to store reader provider and reader unit selection.
    std::shared_ptr<logicalaccess::ReaderConfiguration> readerConfig(new logicalaccess::ReaderConfiguration());
    
    // Set PCSC ReaderProvider by calling the Library Manager which will load the function from the corresponding plug-in
    readerConfig->setReaderProvider(logicalaccess::LibraryManager::getInstance()->getReaderProvider("PCSC"));

}
```

The first line of code starting with std::shared_ptr.. just runs fine. The second line calling the getReaderProvider somehow causes the error. I have no idea where to look for the fix of this error. I tried but no luck. Anyone have any advice?

xaqq, 2020-05-27:
Hello,

I am not able to reproduce the issue.
Is it possible you have multiple version of the library somehow installed and shared library / plugins are incorrectly selected?
If by using a debugger you are able to identify the source of the throw it might tell us more about what is wrong.

roelofd, 2020-05-28:
I only have one version of the library installed, only the 2.2.1 version. I don't have a debugger at this point. I'm installing Clion to maybe find the source of the error in there.

roelofd, 2020-05-28:
> Hello,
> 
> I am not able to reproduce the issue.
> Is it possible you have multiple version of the library somehow installed and shared library / plugins are incorrectly selected?
> If by using a debugger you are able to identify the source of the throw it might tell us more about what is wrong.

I only have one version of the library installed, only the 2.2.1 version. I don't have a debugger at this point. I'm installing Clion to maybe find the source of the error in there.

Maxhy, 2022-10-04:
Build process has been cleaned up, should be fixed anyway now.

closed 2022-10-04 by Maxhy