liblogicalaccess/liblogicalaccess #195: Cannot Build with Visual Studio 2019

Hay1tsme, 2021-03-06:
When I try to do `conan install .. --build=missing` on Windows 10 running VS2019, it fails to build boost because the version of boost that is used doesn't know how to build with 2019. The bootstrap.bat file expects vs142 for Visual Studio 2019, but conan attempts to use vs16, which errors out and fails.

c3rebro, 2021-03-15:
You have to use boost/1.71.0@conan/stable instead.

Don't forget to change the conanfile.py:

requires = 'boost/1.71.0@conan/stable' ...

Hay1tsme, 2021-03-16:
Thank you, that got `conan install` to finish!

Unfortunatly, `conan create` is giving me trouble now. The last line of the log is as follows

`LogicalAccess/2.3.0:
LogicalAccess/2.3.0: ERROR: Package '3c8a64f282955ecee5c3576ee3f3634919e3de37' build failed
LogicalAccess/2.3.0: WARN: Build folder C:\Users\ktroc\.conan\data\LogicalAccess\2.3.0\_\_\build\3c8a64f282955ecee5c3576ee3f3634919e3de37
ERROR: LogicalAccess/2.3.0: Error in build() method, line 97
        cmake.build()
        ConanException: Error 1 while executing cmake --build C:\Users\ktroc\.conan\data\LogicalAccess\2.3.0\_\_\build\3c8a64f282955ecee5c3576ee3f3634919e3de37 --config Release -- /m:16 /verbosity:minimal`

Maxhy, 2021-10-05:
Most compilers do not need these explicit includes that's why it wasn't properly caught. Should be fixed now. Thanks

closed 2021-10-05 by Maxhy