liblogicalaccess/liblogicalaccess #206: __unsupported__ prefix for DesFireEV2

c3rebro, 2022-08-08:
Hello,

Why is this happening? In 1.87 this same chip was recognised as EV1 (But could be identified as EV2 by getVersion cmd). However, Basic functionality was still given. Now, in 2.4.0 it is somewhat not usable anymore.

Is this a readerspecific behaviour? I use OK5321.

c3rebro, 2022-08-10:
Hello @Maxhy 

Further research showed the following error in the log:

```
Cannot get symbol getDESFireEV2Chip on library D:\Seafile\Dokumente Privat\Steven\Projekte\Software\Windows\RFiDGear\RFiDGear\bin\x64\Debug\cps3cards.dll.
2022-Aug-11 00:27:53.553372 - : 	{35}	{logicalaccess::WindowsDynLibrary::getSymbol}	{K:\jenkins_data\lla@2\src\dynlibrary\windowsdynlibrary.cpp}:
Cannot get symbol getDESFireEV2Chip on library D:\Seafile\Dokumente Privat\Steven\Projekte\Software\Windows\RFiDGear\RFiDGear\bin\x64\Debug\desfirecards.dll.
2022-Aug-11 00:27:53.554356 - : 	{35}	{logicalaccess::WindowsDynLibrary::getSymbol}	{K:\jenkins_data\lla@2\src\dynlibrary\windowsdynlibrary.cpp}:
Cannot get symbol getDESFireEV2Chip on library D:\Seafile\Dokumente Privat\Steven\Projekte\Software\Windows\RFiDGear\RFiDGear\bin\x64\Debug\em4102cards.dll.
```

Unfortunately there seems to be no Symbol in the Code. Or i just cannot find the "getDESFireEV2Chip" Method...

Maxhy, 2022-08-19:
EV2 is properly supported with LLA but the current implementation is on private repo only. We don't have a fallback to EV1 chip if EV2 chip factory is not being found, this has to be done by the consuming application for now I guess.
I plan to add another EV2 implementation for the public repo as soon as the chip command set got officially shared on the internet (it's not yet the case and I don't want to be the first one to "leak" it...) so we could reference it without breaking whatever NDA signed by anyone... 😅

c3rebro, 2022-10-11:
Ok. It looks a little „uncommon“ that a visitor got redirected to HID when trying to open some of the islog website links from the wiki. Is the company sold to HID? And is there still a chance to get the latest .net wrapper for open source projects?

Maxhy, 2022-10-11:
Yes ISLOG has been acquired by HID Global mid-2018 ; the ISLOG branding has been stopped since.
Latest .NET wrapper is available on NuGet official repository, be sure to use the version ending by CE for Community Edition.
The related natives binaries can be found on liblogicalaccess-swig repo releases.

pm2403, 2023-08-30:
Hello,
has anything changed about EV2? So far I've been using version 1.83 and EV2 was somehow supported. I converted the program code to version 2.4.0 and now EV2 cannot be handled. Is there any chance to support EV2 in the current version or do I have to resign from this library?

Maxhy, 2023-09-01:
Hi @pm2403.
I have left HID Global company since, but I'm continuing to support LibLogicalAccess library either as a direct contributor, and through my new company (LEOSAC).
I'm planning to release a LLA v3 on coming months with better EV2 support and several cleanups (removal of deprecated plugins, build process...).
Now, LLA v1.83 doesn't have much support of EV2 (not even properly detected). On v.2.4 it is at least properly automatically detected, and you should be able to use EV2 chip just like EV1, what's your issue?

Maxhy, 2024-02-25:
LLA v3 is now released and this issue is fixed.

closed 2024-02-25 by Maxhy