liblogicalaccess/liblogicalaccess #207: Unknown CardService type: MifarePlusStorageSL1 when writing MAD Data to PLUS S chip in SL1

c3rebro, 2022-08-09:
It seems using a StorageService on SL1 Plus Cards is not implemented?

```
var cmd = card.getCommands() as MifareCommands;
var cardService = card.getService(LibLogicalAccess.CardServiceType.CST_STORAGE) as StorageCardService;

try
{
     cardService.writeData(mlocation, aiToUse, aiToWrite, new ByteVector(buffer), CardBehavior.CB_AUTOSWITCHAREA);
}
catch (Exception e)
{
...
}
```

The above throws the mentioned LibLogicalAccessNETCE Exception. I tried with:

```
var aiToUse = new MifareAccessInfo
{
		useMAD = _useMADToAuth, (true)
		keyA = mAKeyToUse,
		keyB = mBKeyToUse

		aiToUse.madKeyA = madAKeyToUse;
		aiToUse.madKeyB = madBKeyToUse;
		aiToUse.madGPB = _madGPB; (0xC1)
};
```

Maxhy, 2022-08-19:
Yeah Mifare Plus are not well supported, see #1. We added such chip support a decade ago but we are not testing it anymore as we used to "not like" this chip for several technical reasons. I guess it could be easily fixed if you are using SL1.

c3rebro, 2022-08-22:
OK, Thank you for replying to all questions J