liblogicalaccess/liblogicalaccess #208: C# Examples don't build

nzgamer41, 2023-04-02:
In the Wiki linked through the README.md file and at the top of the GitHub repo, there is a page for examples using LibLogicalAccessNetCE in C#.

I've tried this today with .NET 6.0 and the very first example won't build. It says that:

The type or namespace name 'IReaderProvider' could not be found
'PCSCReaderProvider' does not contain a constructor that takes 0 arguments
The type or namespacename 'IReaderUnit' could not be found

I've tried this with v2.5.0 and v2.4.2 using Visual Studio 2022 and .NET 6.0 as well as .NET Framework 4.8 as a sanity check and I get the same issue.
Doesn't matter if I add `using LibLogicalAccess;` and `using LibLogicalAccess.Reader;` to the top, it still won't build.

I've included a sample that shows this not working. Only reference I've added is the v2.5.0 NuGet package.

`using LibLogicalAccess.Reader;
using LibLogicalAccess;
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;

namespace MifareTest3
{
    internal class Program
    {
        static void Main(string[] args)
        {
            // Explicitly use the PC/SC Reader Provider.
            IReaderProvider readerProvider = new PCSCReaderProvider();

            // Create the default reader unit. On PC/SC, we will listen on all readers.
            IReaderUnit readerUnit = readerProvider.CreateReaderUnit();

            if (readerUnit.ConnectToReader())
            {
                // DO CARD STUFF HERE
                // DO CARD STUFF HERE
                // DO CARD STUFF HERE

                readerUnit.DisconnectFromReader();
            }
        }
    }
}`

Maxhy, 2023-04-03:
Yes you should now use the NuGet package https://www.nuget.org/packages/LibLogicalAccessNetCE as you did.
The wiki is a bit outdated (was for v1). Not you have to create the reader provider instance through a singleton. Like `var readerProvider = PCSCReaderProvider.createInstance();`
At runtime, do not forget to put the native dlls (https://github.com/liblogicalaccess/liblogicalaccess-swig/releases/download/v2.5.0/liblogicalaccess-swig-bin.zip) side by side to your application.

You can use the following c# sample (up to date for v2) as a reference : https://github.com/liblogicalaccess/liblogicalaccess-samples-csharp

Maxhy, 2023-04-04:
Wiki updated as well.

closed 2023-04-04 by Maxhy