liblogicalaccess/liblogicalaccess #210: Build library and examples without conan?

rm5248, 2023-04-21:
I'm currently trying to build/evaluate liblogicalaccess for use but I'm not sure how to build it.  The available directions are not clear to me, as I have not used conan before.

**What I expect to do to build**
Normal software that I have used doesn't require conan, so you would build like such:
```
$ apt-get install cmake libboost-???-dev
$ mkdir build
$ cd build
$ cmake ..
$ make
$ make install
```

**What I have gotten to**
I've managed to do the following which seems to at least partially work, but the conan install fails:
```
$ pip3 install --upgrade conan==1.59.0
$ mkdir build
$ cd build
$ conan install ..
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++
compiler.version=10
os=Linux
os_build=Linux
[options]
[build_requires]
[env]

cppkcs11/1.1: Not found in local cache, looking in remotes...
cppkcs11/1.1: Trying with 'conancenter'...
ERROR: Failed requirement 'cppkcs11/1.1' from 'conanfile.py (LogicalAccess/2.5.0)'
ERROR: Unable to find 'cppkcs11/1.1' in remotes
```

Two questions with this:
1. If this library works properly, we don't use conan at all on our build and it would be complicated to add that into our existing build, can this be built without conan?
2. How to build the library standalone to simply use the given sample application to determine if it works?  I'm expecting that when the library is built, the sample application will be built with it and immediately usable.

Maxhy, 2023-04-24:
1. Conan is the tool used to manage dependencies and generate proper build environments.
It could be built without Conan with proper manual installation of dependencies, as it was before on the old days of LLA. But as it is not the recommended way you will be on your own here. We should provide pre-built package for Debian at least but such project got deprecated over time as we stopped using it.

2. The error you have is because the dependencies are not available on the official conancenter. This is +/- related to #209. You have to run conan install with `--build=missing` parameter then when using official conancenter repo.

Maxhy, 2023-04-25:
Right, you first need to build [cppkcs11](https://github.com/liblogicalaccess/cppkcs11) as this library is not on conancenter neither.
You can also build LLA without pkcs#11 support if you're not going to use HSMs for key storage, with `-o LLA_BUILD_PKCS=False`

rm5248, 2023-04-26:
Thanks, that gets me a bit farther.  I've been following the [wiki page](https://github.com/liblogicalaccess/liblogicalaccess/wiki/Build-Liblogicalaccess) but it doesn't have much information on it.

The current issue(after installing libpcsclite thru apt) is that now cmake keeps complaining about the wrong number of arguments for SET_TARGET_PROPERTIES:
```
CMake Error at CMakeLists.txt:126 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at plugins/logicalaccess/plugins/iks/CMakeLists.txt:20 (SET_TARGET_PROPERTIES):
  SET_TARGET_PROPERTIES called with incorrect number of arguments.

...(many more lines of the same error)...
```

Maxhy, 2023-09-01:
You can also build the library without IKS support if that's the issue. `-o LLA_BUILD_IKS=False`
IKS is going to be deprecated on the next release anyway.

Maxhy, 2023-12-04:
For LLA v3, IKS feature will be removed and cppkcs11 dependency will be merged to LLA codebase.
This should simplify the overall build process.

Maxhy, 2025-03-16:
Should be fixed since LLA v3.x now.

closed 2025-03-16 by Maxhy