liblogicalaccess/liblogicalaccess #211: No reader provider after update

BertoltLang, 2023-11-21:
Hello,

I recently updated liblogicalaccess from 1.8.0 to 2.5.0 and tried to tie in the external libraries in my setup. 

After the update I now run into the same issue as the user vampik33 described here: https://github.com/liblogicalaccess/liblogicalaccess/issues/194
but I don't understand his solution.

The troublesome code is the following:

```
_readerConfig = std::make_shared<ReaderConfiguration>();
auto libraryManager = LibraryManager::getInstance()->getAvailableReaders();

// Set PCSC ReaderProvider by calling the Library Manager which will load the function
// from the corresponding plug-in
if (std::shared_ptr<ReaderProvider> provider = LibraryManager::getInstance()->getReaderProvider("PCSC")) {
    _readerConfig->setReaderProvider(provider);
} else {
    throw CardReaderException(CardReaderException::UNREADABLE, "Cannot get PCSC provider");
}
```

libraryManager is empty and LibraryManager::getInstance()->getReaderProvider("PCSC") returns only null.

Any type of help or a hint on how to solve this would be much appreciated.

BertoltLang, 2023-12-07:
Hi Maxhy,
thank you for your answer! The config file doesn't seem to be the culprit. I can now see the results of
`auto libraryManager = LibraryManager::getInstance()->getAvailableReaders();`
and it's not empty. (The problem was, that I started my program in debug mode with liblogicalaccess .dlls for release mode)
I still have the problem with 
```
if (std::shared_ptr<ReaderProvider> provider = LibraryManager::getInstance()->getReaderProvider("PCSC")) {
    _readerConfig->setReaderProvider(provider);
} else {
    throw CardReaderException(CardReaderException::UNREADABLE, "Cannot get PCSC provider");
}
```
since
```
PCSCReaderProvider::PCSCReaderProvider()
    : ISO7816ReaderProvider()
{
    d_scc      = 0;
    long scres = SCardEstablishContext(SCARD_SCOPE_USER, nullptr, nullptr, &d_scc);
    if (scres != SCARD_S_SUCCESS)
    {
        char tmpbuf[128];
        memset(tmpbuf, 0x00, sizeof(tmpbuf));
        sprintf(tmpbuf, "Can't establish the context for PC/SC service (%x).",
                static_cast<unsigned int>(scres));
        THROW_EXCEPTION_WITH_LOG(LibLogicalAccessException, tmpbuf);
    }
}
```
in liblogicalaccess-2.5.0\plugins\logicalaccess\plugins\readers\pcsc\pcscreaderprovider.cpp will throw its exception because
`long scres = SCardEstablishContext(SCARD_SCOPE_USER, nullptr, nullptr, &d_scc);`
returns a negative value.
Do you have an idea, what could cause this? I'm having a little trouble to debug it thoroughly.

Maxhy, 2024-01-03:
Ok good.
Mhhh this shouldn't have changed, the logic should be the same between both versions. Are you sure your PC/SC service is properly installed and running (pcsc-lite on Linux)? What's the negative return value?

BertoltLang, 2024-01-25:
Turns out, I just had forgotten something in my deploy process - there were some .dlls missing that prevented me from using the library correctly. While in debugging I had all the .dlls at my disposal. 
Thank you for your help.