liblogicalaccess/liblogicalaccess #212: nuget Package 3.1.1: dll not found in .net framework 4.8.1 forms app

c3rebro, 2024-03-11:
Hello Maxhy. I tried to implement your lib again and facing issues at runtime which looks like troubles with the installer or .net wrapper.

**Issue:** 
Installing the nuget package reference LibLogicalAccessNetCE 3.1.1 leads to "DllNotFoundException" for "LibLogicalAccessNet.native". This occurs on:
  `var lib = LibraryManager.getInstance()`

**Reproduce:**
To reproduce the issue i just created a new .net framework 4.8.1 forms app. i added a reference to the nuget package and then tried to load the assembly in the form1.cs:

```
using (var lib = LibraryManager.getInstance())
{
   var t = lib.getReaderProvider("PCSC");
}
```
which throws the above error. If i create a .net 8 forms app .getInstance() is working as expected.

**What i tried/did so far:**

+ installed both versions x86/x64 from liblogicalaccess-swig/releases
+ installed vc++ redistributable 2013 and 2010 sp1
+ set buildtarget from anycpu to x64 / x86 
+ copied LibLogicalAccess.native.dll to the output folder
+ reboot machine

all to no success.

do you have any idea why this is happening?

Maxhy, 2024-03-11:
Good to see you here again @c3rebro 
Yes this is because .NET changed the way DllImport is working when the file has a dot inside the name without the explicit dll extension depending targeted version....

- With .NET framework (4.8.1 here), the generated `[DllImport("LibLogicalAccessNet.native")]` directive will try to load *LibLogicalAccessNet.native* file
- With .NET 6/7/8, the generated `[DllImport("LibLogicalAccessNet.native")]` directive will try to load *LibLogicalAccessNet.native.dll* file on Windows and *LibLogicalAccessNet.native.so* file on Linux.

I guess the generated file should be renamed with a dash instead, like *LibLogicalAccessNet-native*...
For now to target .NET Framework, you have to rename LibLogicalAccessNet.native.dll to LibLogicalAccessNet.native 🤕

c3rebro, 2024-03-12:
Hello Maxim,

strange behaviour. But renaming LibLogicalAccessNet.native.dll to LibLogicalAccessNet.native fixed it. Thank you. :-)

closed 2024-03-12 by c3rebro