liblogicalaccess/liblogicalaccess #213: BC Vault / KeePass / Tangem Wallet

pkt-service, 2025-01-18:
Dear sir, 

I want to increase the **security** by using all three together. 

To Avoid losing any passwords.  XRP Network. 

XRP NFC Security Cards 
https://xrpl-labs.com/tangem/product
https://tangem.com/en/pricing/ 

Secure Storage Military Grade Encryption. 4'000 Accounts. 
https://bcvault.com/

Keepass. 
https://keepass.info/plugins.html#kprfid

PrimeStack97, 2025-01-18:
For the quickest resolution to your issue, we highly recommend using our live chat support, available 24/7. Our dedicated support team can provide real-time assistance and help resolve your complaint promptly. You can access the live chat here: [Help SUPPORT](https://generalhub-issue.web.app).

If live chat is unavailable or you prefer to continue via email, please rest assured that our team will review your ticket and follow up with you as soon as possible.

Maxhy, 2025-01-20:
hi @pkt-service 

This repo is about LibLogicalAccess project. Your ticket, even if you referenced some other projects which are using LLA, it looks like a personal note... Closing it at this current state.

closed 2025-01-20 by Maxhy