liblogicalaccess/liblogicalaccess #214: writeData() fails with "Data requested but no more data to send" when writing full-sized data to DESFire

c3rebro, 2025-03-28:
Hello @maxhy,

Thanks for continuing the development of this great project! 😍

Unfortunately, it seems that something has changed in the way data is written to DESFire cards after the latest update.

I'm using the Community Edition in a C# .NET 4.8.1 application.

🐞 Problem Description / How to Reproduce:
Create an application and file with a size of 160 bytes.

Use `cmd.readData((byte)_fileNo, 0, (uint)_fileSize, LibLogicalAccess.Card.EncryptionMode.CM_ENCRYPT).ToArray();` to read exactly 160 bytes from the file.

Try to write the exact same data back using writeData():

csharp

```
var cmd = card.getCommands() as DESFireCommands;
cmd.selectApplication((uint)_appID);
cmd.authenticate((byte)_writeKeyNo, aiToWrite.writeKey);
cmd.writeData((byte)_fileNo, 0, new ByteVector(_data), LibLogicalAccess.Card.EncryptionMode.CM_ENCRYPT);
```
Where _data is exactly 160 bytes.

❌ Result:
I get this exception:

`Data requested but no more data to send`
If I add 1 padding byte at the end (161 bytes total), I get the expected error:

`Attempt to read/write data from/to beyond the file's/record's limits / Attempt to exceed the limits of a value file`
However, if I try to write 159 bytes, the write succeeds without any issues.

🤔 What’s Confusing:
Even when writing less than the file size (e.g., 158 bytes), I occasionally still get the "Data requested but no more data to send" error. This suggests the issue may not be strictly size-related, or perhaps chunking behavior has changed.

Could this be a regression or a change in how data framing is handled in writeData()? Is manual chunking or padding now required?

Let me know if you need more details—I’d be happy to provide logs or test further.

Thanks again!

Maxhy, 2025-03-29:
Thanks for the report @c3rebro 
Mhhh this may be indeed a regression related to my refactoring of some commands transmit to avoid code duplicate in EV2 secure communication. I will take care of this, but just to be sure, this was working correctly in v3.2.0?

c3rebro, 2025-03-30:
Thanks for the quick reply, @Maxhy! 🙏

I dug a bit deeper and discovered some details that might help narrow this down:

🧪 Version Check
I was previously using v3.0.1, and the issue did not occur there. I use the liblogicalaccess-swig if that matters.

🔄 Hardware-Specific Behavior
Initially, I didn’t realize I was using two different readers:

✅ Omnikey 5422CL → Works correctly (both in v3.0.1 and v3.3.0)

❌ Sciel SCL3711 → Triggers the "Data requested but no more data to send" error when writing exactly 160 bytes to EV3

So the issue may also be dependent on the PCSC reader model, not just the version.

🧪 Reader Initialization Snippet
This is the method I use to connect to the reader and chip (worked for EV3 in v3.0.1):

```
private async Task<bool> tryInitReader()
{
    return await Task.Run(() =>
    {
        try
        {
            if (readerUnit?.isConnected() == true)
            {
                readerUnit.connect();
                return true;
            }
            else if (readerUnit?.connectToReader() == true)
            {
                if (readerUnit.waitInsertion(Constants.MAX_WAIT_INSERTION))
                {
                    if (readerUnit?.connect() == true)
                    {
                        ReaderUnitName = ReaderUnitName ?? readerUnit.getConnectedName();
                        return true;
                    }
                }
            }
        }
        catch (Exception e)
        {
            return false;
        }
        return false;
    });
}

```

🚫 New Issue After Upgrading to v3.3.0 on EV3 on OK's 5422CL
After switching back to v3.3.0 for more tests, I hit another blocker: I can no longer connect to EV3 chips using this reader setup.
Calling readerUnit?.connect() throws the following exception:

`Wrong Class`
This same call works fine for EV1 and EV2 chips, and it did work for EV3 in v3.0.1. It also works for EV3 chips but only when using the SCL3711.

🔁 Card Switching Oddity
One more thing I noticed:
If I switch cards without calling waitRemoval() (and maybe even when i call it, i have not tested this, i just call card.connect()) between reads, then:

getChipIdentifier() updates correctly with the new card

But getCardType() remains stuck on the old value (e.g., it always shows "EV1")

🔁 Reading Card Info (After connect):

```
card = readerUnit.getSingleChip();

var uidAsString = ByteArrayConverter.GetStringFrom(card.getChipIdentifier().ToArray());
var chipType = (CARD_TYPE)Enum.Parse(typeof(CARD_TYPE), card.getCardType());
```

So maybe there's a caching issue or some initialization that needs to be re-triggered?

Let me know if there's anything I can do to help debug further—I can isolate the behaviors more if needed or share logs.

Thanks again.

Maxhy, 2026-08-17:
Is this issue still valid on your side @c3rebro?
We just identified a similar issue which would only happens if data is 16-byte aligned and over Secure Mode EV2 only. Fix will be part of v3.9.0. It may not be the same issue as yours.

For waitRemoval, it is required on some readers, safer to always call it at the end. If you don't want to wait just pass 1 as the timeout.