liblogicalaccess/liblogicalaccess #215: conanfile does not produce functional package on Windows (VS2022)

YetAnotherGeorge, 2025-03-31:
Hello, i'd appreaciate a bit of help with using the liblogicalaccess on Windows. I'm coming from c# so i'm a bit of a c++ newbie. 

I ran 
git clone https://github.com/liblogicalaccess/cppkcs11; conan export .
git clone https://github.com/islog/liblogicalaccess.git; cd liblogicalaccess;
mkdir build ; cd build ; conan install .. ; conan create ..

Then i used the VS2022 conan extension, with the default conanfile.py as generated by the extension and with the following conandata.yml:
```
requirements:
- boost/1.83.0
- logicalaccess/3.3.1
```

The first build it built things, then the second time it ran, but gave me the error "logicalaccess.dll" missing, the includes were handled ok tho.

Then i manually added the dll with a post-build event (xcopy /Y /D ... "$(TargetDir)") for each missing dll.

Here is the output of one of the code samples you provided:

[image]

I have a smartcard reader that registers 5 PC/SC devices in device manager and i think this library would be very helpful for me to test the device.

Please help, and thanks

YetAnotherGeorge, 2025-04-02:
Hello, I looked for missing pcscreaders.dll, and the other files you mentioned, and the error happened with them present.

Here are the steps I took:
1. conan remove "*"; conan cache clean "*"
2. Script:
```
mkdir C:\LIBCPP; cd C:\LIBCPP;

# liblogicalaccess-cppkcs11
git clone https://github.com/liblogicalaccess/cppkcs11
cd .\cppkcs11
conan export .
cd ..\

# liblogicalaccess
git clone --branch 3.3.0 https://github.com/liblogicalaccess/liblogicalaccess.git
cd liblogicalaccess; mkdir build; cd build;
conan install .. --build=cppkcs11/1.2 --build=gtest/1.15.0 --build=openssl/3.4.1
conan create ..

# Output of conan list now: (conan remove "*"; conan cache clean "*" was run before script)
# Found 10 pkg/version recipes matching * in local cache
# Local Cache
#   b2
#     b2/5.2.1
#   boost
#     boost/1.83.0
#   bzip2
#     bzip2/1.0.8
#   cppkcs11
#     cppkcs11/1.2
#   gtest
#     gtest/1.15.0
#   logicalaccess
#     logicalaccess/3.3.0
#   nlohmann_json
#     nlohmann_json/3.11.3
#   openssl
#     openssl/3.4.1
#   strawberryperl
#     strawberryperl/5.32.1.1
#   zlib
#     zlib/1.3.1
```

3. I created a fresh C++ Console App from VS2022
4. I added the following to conandata.yml:
```
requirements:
- boost/1.83.0
- openssl/3.4.1
- nlohmann_json/3.11.3
- zlib/1.3.1
- logicalaccess/3.3.0
```
At first i installed boost/1.83.0 to make the extension generate the template conandata.yml and conanfile.py, but adding all items except logicalaccess/3.3.0 i don't think is necessary

5. ctrl+f5 (start without debugging) -> Conan extension started building stuff
6. run ctrl+f5 again
7. Exe complains no logicallaccess libs present
8. I added the following Post-Build events:
```
xcopy /Y /D "C:\Users\georg\.conan2\p\b\logic40e293f73f300\p\bin\*" "$(TargetDir)"
xcopy /Y /D "C:\Users\georg\.conan2\p\b\opensedae278dd3e0d\p\bin\*" "$(TargetDir)"
```
I think that's where conan puts build files, but idk if these dlls are for the same configuration as my project (x64)
9. Now the following list of files is present in my debug dir:
```
TestConan5.exe      epasscards.dll         iso15693cards.dll            mifarecards.dll            samavcards.dll
TestConan5.pdb      felicacards.dll        iso7816cards.dll             mifarepluscards.dll        seoscards.dll
basic.exe           generictagcards.dll    iso7816readers.dll           mifareultralightcards.dll  smartframecards.dll
c_rehash.pl         gunneboreaders.dll     keyboardreaders.dll          ok5553readers.dll          stidstrreaders.dll
cps3cards.dll       icode1cards.dll        legicprimecards.dll          openssl.exe                stmlri512cards.dll
deisterreaders.dll  icode2cards.dll        libcrypto-3-x64.dll          osdpreaders.dll            tagitcards.dll
desfirecards.dll    indalacards.dll        libssl-3-x64.dll             pcscreaders.dll            topazcards.dll
elatecreaders.dll   infineonmydcards.dll   llacommon.dll                pkcscryptounified.dll      twiccards.dll
em4102cards.dll     islogkbdhook64.exe     logicalaccess-cryptolib.dll  proxcards.dll              yubikeycards.dll
em4135cards.dll     islogkbdhooklib64.dll  logicalaccess.dll            proxlitecards.dll
```

10. I don't get any more file errors but still read access violation:
[image]

I'm not running it as admin, i have a HID ATR220 connected via usb so there are pc/sc devices.
Would it help if i were to give you the vs2022 project? Can i find prebuild dlls for windows for logicalaccess?

Many thanks

Maxhy, 2025-04-02:
What you do doesn't looks wrong.
What if you run the basic.exe here from your output directory, does it list correctly all reader providers / card technologies? Can you use the PCSC reader provider from it and just wait for card insertion?
Also, can you run your program with the working directory at the build output directory (using cmd, cd and run outside vs2022 just for testing)?

I have added pre-built dlls as a raw zip archive (to not need msi/msm/nuget) to the swig releases, see https://github.com/liblogicalaccess/liblogicalaccess-swig/releases/download/3.3.0/liblogicalaccess-swig-bin.zip Simply ignore LibLogicalAccessNet.native.dll which is only needed for the generated C# wrapper that you don't use here)

YetAnotherGeorge, 2025-04-03:
When i run ./basic.exe it works fine:
```
Available reader plug-ins (8):
        Deister Elatec Gunnebo Keyboard OK5553 OSDP PCSC STidSTR
Available card plug-ins (41):
        CPS3 DESFire DESFireEV1 DESFireEV2 DESFireEV3 EM4102
        EM4135 EPass FeliCa GenericTag iCode1 iCode2 Indala
        InfineonMYD ISO15693 ISO7816 LegicPrime
        Mifare1K Mifare4K MifarePlus_SL0_4K MifarePlus_SL0_2K
        MifarePlus_SL1_4K MifarePlus_SL1_2K MifarePlus_SL3_4K MifarePlus_SL3_2K
        MifarePlusS MifarePlusX MifareUltralight MifareUltralightC Prox
        ProxLite SAM_AV1 SAM_AV2 SAM_AV3 Seos SmartFrame StmLri512 TagIt
        Topaz Twic Yubikey
Please type the reader plug-in to use:
PCSC
5 readers on this system.
Please select index of the reader unit to use:
        0. Access IS NFC0 0
        1. Access IS SMC0 0
        2. Access IS SMC1 0
        3. Access IS SMC2 0
        4. Access IS SMC3 0
0
Waiting 15 seconds for a card insertion...
Time start : 1743680805
Card inserted on reader "Access IS NFC0 0".
Card type: DESFireEV1
Card Serial Number : ......
Logical automatic card removal in 15 seconds...
Card removed.
Time start : 1743680809
Card inserted on reader "Access IS NFC0 0".
Card type: DESFireEV1
Card Serial Number : .....
Logical automatic card removal in 15 seconds...
Card removed.
Time start : 1743680821
No card inserted.
Time start : 1743680836
```

Maxhy, 2025-04-12:
Mhhh I don't see your issue here.
What if you copy/past your code to basic.cpp temporary and try to build liblogicalaccess/run basic with this code? Just to be sure it is not related to this specific code on your environment, even if there is no reason and that it works on my side...
Which conan profile are you using when you build? (or what is your default profile as it doesn't looks like you specify any)?

YetAnotherGeorge, 2025-04-12:
When i'll get back to the office i'll try what you suggested and update you, so when you build the repo i gave you, it works fine? The same code? 

Also here is what conan profile list and conan profile show give me on the pc in question:
```
PS C:\Users\georg> conan profile list
Profiles found in the cache:
default
PS C:\Users\georg> conan profile show
Host profile:
[settings]
arch=x86_64
build_type=Release
compiler=msvc
compiler.cppstd=20
compiler.runtime=dynamic
compiler.runtime_type=Release
compiler.version=194
os=Windows

Build profile:
[settings]
arch=x86_64
build_type=Release
compiler=msvc
compiler.cppstd=20
compiler.runtime=dynamic
compiler.runtime_type=Release
compiler.version=194
os=Windows

```

YetAnotherGeorge, 2025-04-15:
Ok, now I did not change the conan profile whatsoever after running conan detect, with the same result:
```
PS C:\Users\georg> conan profile show
Host profile:
[settings]
arch=x86_64
build_type=Release
compiler=msvc
compiler.cppstd=14
compiler.runtime=dynamic
compiler.runtime_type=Release
compiler.version=194
os=Windows

Build profile:
[settings]
arch=x86_64
build_type=Release
compiler=msvc
compiler.cppstd=14
compiler.runtime=dynamic
compiler.runtime_type=Release
compiler.version=194
os=Windows
```

Program Output (regardless of if i run as admin or not):
```
Start
Reader configuration object created.
Reader provider set to PCSC.

D:\Backup\CS.Programming\Languages\C++\TestLibLogicalAccess\x64\Debug\TestLibLogicalAccess.exe (process 9452) exited with code -1073741819 (0xc0000005).
Press any key to close this window . . .
```
It's the same error:
```
Exception thrown: read access violation.
std::shared_ptr<logicalaccess::ReaderProvider>::operator-><logicalaccess::ReaderProvider,0>(...) returned nullptr.
```